hku-mars/r3live #7: The 3D-printable handheld device for R2LIVE and R3LIVE

OliverShaoPT, 2021-11-10:
Hello @ziv-lin @XW-HKU,

Thanks for sharing your great works. I notice that you have kindly release the 3D-printable handheld device in Loam-livox repository.  If possible, can you please also share the the 3D-printable handheld device for R2LIVE and R3LIVE?

Best regards

ziv-lin, 2021-11-10:
As claimed in our paper, the 3D FDM printable handheld device will be released at https://github.com/ziv-lin/rxlive_handheld, before the date Dec 31, 2021.

OliverShaoPT, 2021-11-11:
> As claimed in our paper, the 3D FDM printable handheld device will be released at https://github.com/ziv-lin/rxlive_handheld, before the date Dec 31, 2021.

Another small question, how do you deal with the time synchronization between Livox Avia and FLIR camera? Is NTP enough for R3LIVE?  (As the BFS-u3-13y3c camera doesn't support the PTP.)

ziv-lin, 2021-11-11:
For the camera sensor, we do not make any hardware synchronization between LiDAR and camera sensor, the time-offset between IMU (LiDAR built-in) and LiDAR is online estimated by our algorithm (both R2LIVE and our new work R3LIVE).

OliverShaoPT, 2021-11-11:
> For the camera sensor, we do not make any hardware synchronization between LiDAR and camera sensor, the time-offset between IMU (LiDAR built-in) and LiDAR is online estimated by our algorithm (both R2LIVE and our new work R3LIVE).

Thanks. I also found that your algorithm considers the time-offset between IMU and camera in the state vector.

closed 2021-11-11 by OliverShaoPT