hku-mars/r3live #8: Point cloud coloring

aditdoshi333, 2021-11-14:
Hello,

I am really excited to try out new pipeline R3live with livox avia. It looks really robust. I need one favour from your end. Currently for a college project I am using livox avia with fastlio 2. But I need rgb colored point cloud or mesh. I tried modifying fastlio code by coloring every livox point cloud msg with rgb image in pre-processing stage. For calibration between lidar and camera I am using (https://github.com/hku-mars/livox_camera_calib). But the result is very noisy, other than rgb points there are many white and black points. I have already made sure that camera and livox are in sync. Can you throw some light on your approach of coloring point or texturing mesh? I know you are going to release the code by 31 dec but I have to submit early. If possible please throw some light for the same. 

Thank you

Camilochiang, 2021-11-14:
Hei @aditdoshi333 . I guess that if you are in a hurry you could give it a try to this [repository:](https://github.com/Livox-SDK/livox_camera_lidar_calibration), for testing your colors , not for calibration as you already did. 

- You could use only the colorLidar.launch to first test that you are getting a good point cloud texture in a fix status. Then you can  update the values with bayesian update as they propose in R3Live. 
- Also what do you mean with noise? You can remove all points that have no color assigned. Did you limit the point cloud of fast lio 2 to only the intersection with the camara FOV?
- Share some screenshots to have a look!

aditdoshi333, 2021-11-14:
Hey @Camilochiang ,

Thanks for the input. I am not sure about the bayesian update method in r3live. Let me share the coloring code with you and result as well.

`
auto rgb_image = Measures.rgb.back();
                cv::Mat src_img = cv_bridge::toCvShare(rgb_image, "bgr8")->image; //image_raw is the image we got
                cv::Mat view, rview, map1, map2;
                cv::Size imageSize = src_img.size();
                
                int image_rows = src_img.rows;
                int image_cols = src_img.cols;
                                vector<vector<int>> color_vector;
                color_vector.resize(image_rows*image_cols);
                for (int v = 0; v < image_rows; ++v) {
                    for (int u = 0; u < image_cols; ++u) {

                        auto color_inst =  (cv::Vec3b) src_img.at<cv::Vec3b>(v, u);
                        // std::cout<<"COLORS::"<<color_inst<<std::endl;
                        // for .bmp photo, the 3 channels are BGR
                        // std::cout<<"BNumbers"<<(int)color_inst[0]<<std::endl;
                        color_vector[v*image_cols + u].resize(3);
                        
                        color_vector[v*image_cols + u][0] = color_inst[2];
                        color_vector[v*image_cols + u][1] = color_inst[1];
                        color_vector[v*image_cols + u][2] = color_inst[0];
                    }
                }

                

                // color_vector.resize(row*col);
                for (int nidx = 0; nidx < Measures.lidar->points.size(); nidx++){

                    float UV[2] = {0, 0};
                    // void getUV(const cv::Mat &matrix_in, const cv::Mat &matrix_out, float x, float y, float z, float* UV) {
                    double matrix3[4][1] = {Measures.lidar->points[nidx].x, Measures.lidar->points[nidx].y, Measures.lidar->points[nidx].z, 1};
                    cv::Mat coordinate(4, 1, CV_64F, matrix3);
                    
                    // calculate the result of u and v
                    cv::Mat result = intrisicMat*extrinsicMat_RT*coordinate;
                    float u = result.at<double>(0, 0);
                    float v = result.at<double>(1, 0);
                    float depth = result.at<double>(2, 0);
                    
                    UV[0] = u / depth;
                    UV[1] = v / depth; 
                    
                    int u_internal = int(UV[0]);
                    int v_internal = int(UV[1]);

                    int32_t index = v_internal*image_cols + u_internal;
                    std::uint8_t r = 0, g = 255, b = 0;
                    if (index < image_rows*image_cols && index >= 0) {
                        r = color_vector[index][0];
                        g = color_vector[index][1];
                        b = color_vector[index][2];
                        std::uint32_t rgb = ((std::uint32_t)r << 16 | (std::uint32_t)g << 8 | (std::uint32_t)b);
                        Measures.lidar->points[nidx].rgb = *reinterpret_cast<float*>(&rgb);
                    }
                    
                }
            }

[image]
[image]
[image]

`

Thanks for guiding. Appreciated

Camilochiang, 2021-11-14:
@aditdoshi333 Can you please add a picture / rosbag ? I could have a look

Camilochiang, 2021-11-14:
Also. I dont see that you are updating the values at all in your code...

aditdoshi333, 2021-11-14:
Hello,

Thanks a lot for putting efforts. 
Sample bag file link: https://drive.google.com/file/d/1KfcpybsyyWcvANpGxAU4fjFZ7WNf48KT/view?usp=sharing

Can you please elaborate on updating values? In preprocessing I am applying the above code block for every sync rgb image with lidar msg. 

Thank a lot again.

Camilochiang, 2021-11-15:
Well here i see two problems:
- did you configure the camera for your actual setup? is quite dark down there and you could increase the light so is not weird that you have only "black points". See an actual screenshot of your video..
- Are you uploading the BRG values of each point? or just replacing them with the last view?
[image]

ziv-lin, 2022-01-11:
This issue is closed for no more updates.
Thanks~

closed 2022-01-11 by ziv-lin