hku-mars/r3live #11: 您好，请问您在测试的时候有遇到过这个问题吗

ToutDonner, 2022-01-01:
================================================================================REQUIRED process [r3live_mapping-3] has died!
process has died [pid 7193, exit code -11, cmd /home/crz/SLAM/R3Live/devel/lib/r3live/r3live_mapping __name:=r3live_mapping __log:=/home/crz/.ros/log/b600fbac-6ad9-11ec-b8ae-2cf05d2b732e/r3live_mapping-3.log].
log file: /home/crz/.ros/log/b600fbac-6ad9-11ec-b8ae-2cf05d2b732e/r3live_mapping-3*.log
Initiating shutdown!
[================================================================================

编译过后运行r3live_bag.lanuch，使用您提供的bag运行就会提示出这个错误，我的ros是melodic

cybaol, 2022-01-01:
我也是
光流模块内存分配失败，然后r3live_mapping就shutdown了。

ToutDonner, 2022-01-01:
> 我也是 光流模块内存分配失败，然后r3live_mapping就shutdown了。

那您现在解决了吗，是我的系统有问题吗

ziv-lin, 2022-01-01:
你opencv是什么版本？还有你电脑的内存多大？

ziv-lin, 2022-01-01:
你能否把这个程序打印截图一下给我？
[image]

cybaol, 2022-01-01:
[image]

ziv-lin, 2022-01-01:
你的内存才7.51Gb啊。。。建议分配下swap试下

ziv-lin, 2022-01-01:
你参考下这个教程，你添加下swap（虚拟内存）试试？
https://www.cnblogs.com/bigben0123/p/13066136.html

slam-art, 2022-01-02:
> 你能否把这个程序打印截图一下给我？ [image]

大佬你好，16GB物理内存和2GB虚拟内存，也是报这个错误，r3live是对内存大小要求很高吗？多少能正常运行呢？

ziv-lin, 2022-01-02:
16GB 肯定是够的，一般来说空余内存大于12GB就行，你这问题有点奇怪，能否详细阐述下？
关于内存占用问题，是因为在程序里面一开始我为了避免std constainer频繁动态分配内存导致整体程序整体性能低下，所以我在一开始就reserve好一个比较大的buffer，如：pointcloud_rgbd.cpp 中的168行这里，以及光流模块image的buffer的allocate。
https://github.com/hku-mars/r3live/blob/bd8d8afe956a3e98f58d63856570c69cbf01915c/r3live/src/rgb_map/pointcloud_rgbd.cpp#L168

ToutDonner, 2022-01-02:
> 你能否把这个程序打印截图一下给我？ [image]
[image]

您好，这个是我的配置信息

ToutDonner, 2022-01-02:
> 你能否把这个程序打印截图一下给我？
[image]
您好，这个是我的配置信息

ziv-lin, 2022-01-02:
你尝试把pointcloud_rgbd.cpp 中的168行里面的1e9改小一点试试，例如1e7之类的？如果是这个问题的话，我提交下代码修下这个bug

ziv-lin, 2022-01-02:
我在DJI-manifold (8GB RAM+ 8 GB swap) 上面验证了这个bug，一开始在16GB内存的机子上的确有bad malloc的问题，1e9一开始确实分配有点大了，我改成：
m_rgb_pts_vec.reserve( 1e8 ); 
就可以了。
这是改之前的
[image]
这是改之后的:
[image]
我稍等把修复推上去

ToutDonner, 2022-01-02:
> 你尝试把pointcloud_rgbd.cpp 中的168行里面的1e9改小一点试试，例如1e7之类的？如果是这个问题的话，我提交下代码修下这个bug

我修改过后还是不可以，但是我可以用r3live_bag.launch运行自己采集的数据（只使用了一台avia），仍然可以完成建图，内存一直维持在4.8GB左右，用fastlio也可以运行您google网盘上传的数据。

ziv-lin, 2022-01-02:
我擦，这就奇怪了，我DJI manifold 8GB RAM + 8GB swap都可以跑额。。。

ziv-lin, 2022-01-02:
这是我在DJI-manifold2 (8GB RAM+ 8 GB swap) 上的测试结果
[image]

ToutDonner, 2022-01-02:
> 我擦，这就奇怪了，我DJI manifold 8GB RAM + 8GB swap都可以跑额。。。

我在编译的过程中爆出一个这样的waring，会不会是这里有问题。或者是我的opencv版本不兼容？但是如果是这里有问题为啥我自己的数据又可以运行
[image]
这个是我用r3live_bag.launch运行自己的数据的一些截图
[image]

ziv-lin, 2022-01-02:
这是因为你没有图像导致的，你看camera-frame那儿都是0

ziv-lin, 2022-01-02:
你命令行窗口输入这个我看下：
```
pkg-config --modversion opencv
```

ToutDonner, 2022-01-02:
[image]

ziv-lin, 2022-01-02:
你刚才截图给我的时候，你编译的时候openCV版本是4.5.4，可是为什么你的运行环境里面是4.5.3呢？
[image]

ziv-lin, 2022-01-02:
另外你有没有按照readme里面说的按照ros的相关packages？
sudo apt-get install ros-XXX-cv-bridge ros-XXX-tf ros-XXX-message-filters ros-XXX-image-transport ros-XXX-image-transport*
如果你没有安装ros-XXX-image-transport*，有可能你是没有解压解compressed_image的相关模块

ToutDonner, 2022-01-02:
> 另外你有没有按照readme里面说的按照ros的相关packages？ sudo apt-get install ros-XXX-cv-bridge ros-XXX-tf ros-XXX-message-filters ros-XXX-image-transport ros-XXX-image-transport* 如果你没有安装ros-XXX-image-transport*，有可能你是没有解压解compressed_image的相关模块

readme里面提到的我有安装，您看一下有问题没。opencv这个问题我再研究一下，我做系统的时候就是装的4.5.4，以前我查询过opencv显示确实也是4.5.3，但是一直没有什么错误，我也就没在意
[image]

ziv-lin, 2022-01-02:
你使用下rqt_image_view, 看能否看到compressed image？就是rosbag play的你能否看到图像？如果能看到，说明你的解压模块没有问题

ziv-lin, 2022-01-02:
你看下能否查看到图像？像这样的：
[image]

ToutDonner, 2022-01-02:
> 你看下能否查看到图像？像这样的： [image]

可以的
[image]

ziv-lin, 2022-01-02:
看起来是对的，然后你跑我的example报什么错误来着？

ToutDonner, 2022-01-02:
> 看起来是对的，然后你跑我的example报什么错误来着？
每次play bag就这样，我现在换一台电脑是以下
[image]

ziv-lin, 2022-01-02:
额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？

ToutDonner, 2022-01-02:
> 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？

恩恩，好的没问题，另外想问一下您后续的工作中会考虑将r3live部署到arm架构下进行测试吗

ziv-lin, 2022-01-02:
我短时间内不大会考虑，比较类似openCV之类的东西移植到arm上面肯定是一堆兼容性问题，扎进这一堆工程细节一堆bug上就没完没了，先考虑下怎么样才能毕业吧。。。头秃。。。

slam-art, 2022-01-02:
> 

> > 你尝试把pointcloud_rgbd.cpp 中的168行里面的1e9改小一点试试，例如1e7之类的？如果是这个问题的话，我提交下代码修下这个bug
> 
> 我修改过后还是不可以，但是我可以用r3live_bag.launch运行自己采集的数据（只使用了一台avia），仍然可以完成建图，用fastlio也可以运行您google网盘上传的数据。

> 16GB 肯定是够的，一般来说空余内存大于12GB就行，你这问题有点奇怪，能否详细阐述下？ 关于内存占用问题，是因为在程序里面一开始我为了避免std constainer频繁动态分配内存导致整体程序整体性能低下，所以我在一开始就reserve好一个比较大的buffer，如：pointcloud_rgbd.cpp 中的168行这里，以及光流模块image的buffer的allocate。 [...]

改成1e8和1e7都不行，我安装的是最新的OpenCV 4.5.5，刚发现编译的时候有以下警告：

> /usr/bin/ld: warning: libopencv_imgcodecs.so.3.2, needed by /opt/ros/melodic/lib/libcv_bridge.so, may conflict with libopencv_imgcodecs.so.405
> /usr/bin/ld: warning: libopencv_imgproc.so.405, needed by /home/wb/3rd/opencv-4.5.5/build/lib/libopencv_imgcodecs.so.4.5.5, may conflict with libopencv_imgproc.so.3.2

可能跟这个有关吧？

ziv-lin, 2022-01-02:
关于内存预分配的问题，我提交的修改如下，当检测到内存小于12Gb的时候，之分配1e8个点的buffer，否则就分配多一点的内存(1e9)
[image]

ziv-lin, 2022-01-02:
> 改成1e8和1e7都不行，我安装的是最新的OpenCV 4.5.5，刚发现编译的时候有以下警告：
>
>
> 可能跟这个有关吧？

额，能否确保下这2者一样？编译时用的库和实际运行的库不match的话总会出现一些奇怪的问题。

ToutDonner, 2022-01-03:
> 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？

您好，问题解决了，应该就是版本没有管理好，我重新装了一个opencv3.4.16，修改了cmake文件。
[image]

ziv-lin, 2022-01-03:
可以可以，大佬666，感谢你分享你的解决过程

ToutDonner, 2022-01-03:
> 可以可以，大佬666，感谢你分享你的解决过程

不敢当不敢当，谢谢你帮我逐个排查问题

closed 2022-01-03 by ToutDonner

h-wata, 2022-01-05:
I am facing the same problem. 
I heard that you modified cmake, can you tell me which part you edited? 
My current environment is as follows. 

[image]

[image]

> > 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？
> 
> 您好，问题解决了，应该就是版本没有管理好，我重新装了一个opencv3.4.16，修改了cmake文件。 [image]

h-wata, 2022-01-05:
I downgraded the OpenCV version from 4.5.1 to 3.4.0 as shown below and rebuilt and it worked. 

[image]

> I am facing the same problem. I heard that you modified cmake, can you tell me which part you edited? My current environment is as follows. ROS:melodic
> 
> [image]
> 
> [image]
> 
> > > 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？
> > 
> > 
> > 您好，问题解决了，应该就是版本没有管理好，我重新装了一个opencv3.4.16，修改了cmake文件。 [image]

ziv-lin, 2022-01-05:
Nice!!! It seems that I need to add this point to ReadME.md for helping other users to address the same problem.

ToutDonner, 2022-01-05:
> 我将 OpenCV 版本从 4.5.1 降级到 3.4.0，如下所示，并重新构建，它工作正常。 谢谢。
> 
> [image]
> 
> > 我面临同样的问题。听说你修改了cmake，能告诉我你修改了哪个部分吗？我现在的环境如下。ROS:旋律
> > [image]
> > [image]
> > > > ，我也看不出问题了，不你先保证下openCV的编译版本和运行版本？不排除这个原因导致的，openCV的问题也一直折腾了我。又或者换一个干净的环境重试下？有结果了和我同步下？
> > > 
> > > 
> > > 你好，问题解决了，应该是版本没有管理好，我重新装了一个opencv3.4.16，修改了cmake文件。 [image]

Sorry,I was taking an exam just now.My opencv version have something wrong,but yours are not.May be opencv 4.xx doesn't work in r3live?

dandelion-chu, 2022-05-12:
> > 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？
> 
> 您好，问题解决了，应该就是版本没有管理好，我重新装了一个opencv3.4.16，修改了cmake文件。 [image]

你好，我也遇到了一样的问题，我用的melodic，安装ros的时候自动安装了opencv3.2，然后想换到opencv4.5.3的，但是配置一直有问题，我发现链接的时候还是链接到了/usr/lib/x86_64-linux-gnu目录下面的opencv的so库了，修改cv_bridge下面的cmake之后，不找这个路径了，但是cv_bridge开始报错找不到函数定义了，能请教下你是怎么更新的opencv版本嘛？谢谢哈~~

heisetiantang, 2022-12-06:
你好我也同样遇到了类似的问题,之前系统自动安装的opencv3。2不符合标准，我再次装了一个opencv4.5.3,编译可以成功，但是运行数据集就报错，希望能得到指导，本人功力尚浅，对于这些路径，配置不太擅长，希望能说说具体的解决方式

heisetiantang, 2022-12-06:
[image]
[image]

dandelion-chu, 2022-12-06:
> [image] [image]

我是参考这两个网页连接中的内容配置的，你可以研究下
https://blog.csdn.net/qq_36758461/article/details/119819161 
https://blog.csdn.net/bigdog_1027/article/details/79092263

ToutDonner, 2022-12-06:
> [image] [image]

你这样的报错看不出来是哪里有问题，可以查一下rqt_image_view，要不你就单独启动node再看具体的错误

heisetiantang, 2022-12-06:
你好，多谢指点，我尝试了rqt_image命令发现没有图像，后面发现opencv的运行版本是直接在cmaklist中直接制定4。5。3，通过命令查询得到的版本是3。2，不知道是不是这个问题

[image]

dandelion-chu, 2022-12-06:
> 

不客气，应该就是版本不一致导致的，我之前遇到的是跑起来会挂掉，统一版本后才能正常运行，但是那个三维重建mesh的节点运行还是有问题，后面没再研究mesh

pereira90-ai, 2024-12-20:
> > 额，我也看不出问题了，要不你先保证下openCV的编译版本和运行版本一致？不排除是不是这个原因导致的，openCV的问题也一直折腾了我很久。又或者换一个干净的环境重新试下？有结果了和我同步下？
> 
> 恩恩，好的没问题，另外想问一下您后续的工作中会考虑将r3live部署到arm架构下进行测试吗

Hello there.

Could you please let me know how to downgrade opencv version of the ros?

I followed ROS installation guide to install ros-noetic and current opencv version is 4.5.4 

How can I downgrade opencv 4.5.4 into the lower version?