hku-mars/r3live #12: Compilation failed in ROS Melodic

viniciusvidal2, 2022-01-02:
Hello everyone,

just faced a compilation failure in ROS Melodic regarding conflict between system libraries.

In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/src/r3live.cpp:69:
/usr/include/flann/ext/lz4.h:196:57: error: conflicting declaration ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/./src/loam/include/common_lib.h:13,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/src/r3live.cpp:59:
/usr/include/lz4.h:196:57: note: previous declaration as ‘typedef struct LZ4_stream_t LZ4_stream_t’
 typedef struct { long long table[LZ4_STREAMSIZE_U64]; } LZ4_stream_t;
                                                         ^~~~~~~~~~~~
In file included from /usr/include/flann/util/serialization.h:9:0,
                 from /usr/include/flann/util/matrix.h:35,
                 from /usr/include/flann/flann.hpp:41,
                 from /usr/include/pcl-1.8/pcl/kdtree/flann.h:50,
                 from /usr/include/pcl-1.8/pcl/kdtree/kdtree_flann.h:45,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/src/r3live.cpp:69:
/usr/include/flann/ext/lz4.h:249:72: error: conflicting declaration ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~
In file included from /opt/ros/melodic/include/roslz4/lz4s.h:38:0,
                 from /opt/ros/melodic/include/rosbag/stream.h:46,
                 from /opt/ros/melodic/include/rosbag/chunked_file.h:46,
                 from /opt/ros/melodic/include/rosbag/bag.h:41,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/./src/loam/include/common_lib.h:13,
                 from /home/vinicius/uwo_ws/src/r3live/r3live/src/r3live.cpp:59:
/usr/include/lz4.h:249:72: note: previous declaration as ‘typedef struct LZ4_streamDecode_t LZ4_streamDecode_t’
 typedef struct { unsigned long long table[LZ4_STREAMDECODESIZE_U64]; } LZ4_streamDecode_t;
                                                                        ^~~~~~~~~~~~~~~~~~
r3live/r3live/CMakeFiles/r3live_mapping.dir/build.make:62: recipe for target 'r3live/r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o' failed

A solution can be found in the following link: https://github.com/ethz-asl/lidar_align/issues/16

ziv-lin, 2022-01-02:
Try this?
```
sudo mv /usr/include/flann/ext/lz4.h /usr/include/flann/ext/lz4.h.bak
sudo mv /usr/include/flann/ext/lz4hc.h /usr/include/flann/ext/lz4.h.bak

sudo ln -s /usr/include/lz4.h /usr/include/flann/ext/lz4.h
sudo ln -s /usr/include/lz4hc.h /usr/include/flann/ext/lz4hc.h
```

ziv-lin, 2022-01-02:
It seems that you have solved your problem hahaha~

viniciusvidal2, 2022-01-02:
Thank you, these commands worked for me too like a charm! Now no one will get stuck with this in the future.

Camilochiang, 2022-01-02:
@viniciusvidal2 close the issue if is solved ;)

closed 2022-01-02 by viniciusvidal2