hku-mars/r3live #15: catkin_make  error

hewujunTDT, 2022-01-03:
error: invalid initialization of reference of type ‘const cv::ParallelLoopBody&’ from expression of type ‘render_pts_in_voxels_mp(std::shared_ptr<Image_frame>&, std::unordered_set<std::shared_ptr<RGB_Voxel> >*, const double&)::<lambda(const cv::Range&)>’
                           { thread_render_pts_in_voxel(r.start, r.end, img_ptr, &g_voxel_for_render, obs_time); });

ziv-lin, 2022-01-03:
Such errors are caused due to the lower OpenCV version (less than 3.3). Please see Section 3.4 in our ReadMe.md

hewujunTDT, 2022-01-03:
> Such errors are caused due to the lower OpenCV version (less than 3.3). Please see Section 3.4 in our ReadMe.md

thanks

ziv-lin, 2022-01-11:
This issue is closed for no more updates.
Thanks~

closed 2022-01-11 by ziv-lin