hku-mars/r3live #16: Transformation matrix

aditdoshi333, 2022-01-03:
Hello,

Thanks for sharing the code. In the config YAML file, there is extrinsic between camera and lidar, I want to know that the matrix is the camera to lidar or lidar to the camera? 

And is there any reference for fine calibration between camera and lidar?

Thank you

ziv-lin, 2022-01-03:
You can get the detail about the relation among senors by referring to our paper, which have give the detail definitions of these frame.

aditdoshi333, 2022-01-03:
Yes @ziv-lin I read details regarding hardware setup in your paper. But I am not able to find which transformation matrix is being used?

Camera to lidar or lidar to camera 
Thank you

ziv-lin, 2022-01-03:
See R2LIVE section III.B, our extrinsic denote the sensor frame w.r.t. IMU frame.
[image]

aditdoshi333, 2022-01-03:
Okay thanks for the reply. But I am getting unexpected output.
[image]

The yellow color is of walls and pillars but on the map, it is showing on the ceiling. I am using a transformation matrix wrt IMU frame.  Any clue what is happening here?

Thank you

aditdoshi333, 2022-01-03:
Hello,

I read your paper every transformation is wrt to imu frame. I have a calibration matrix between lidar and RGB. But not with imu are you using some static matrix between lidar and imu because there is no such information in the config file. I am using livox avia with your custom driver. 

If you are using such a matrix I can use that to get transformation between RGB and IMU.

Thank you

aditdoshi333, 2022-01-04:
As suggested the following my hardware and software configuration
[image]

ziv-lin, 2022-01-04:
> Okay thanks for the reply. But I am getting unexpected output. [image]
> 
> The yellow color is of walls and pillars but on the map, it is showing on the ceiling. I am using a transformation matrix wrt IMU frame. Any clue what is happening here?
> 
> Thank you

It seems that the calibration between the camera and IMU is too bad, did you have a good calibration of your camera intrinsic, and the extrinsic between IMU and camera?

aditdoshi333, 2022-01-04:
Hello @ziv-lin ,

I think calibration is good enough because I am using the same matrix in fastlio2 (modified) for coloring point cloud and doing a decent job. But I think the way I am putting the matrix there is some with that. Is there any repo or something you can suggest for imu and rgb camera?

ziv-lin, 2022-01-04:
For calibrating the intrinsic of the camera, you can use the tools provided by OpenCV or Matlab. And to perform the LiDAR-camera calibration, I recommend you to use this repo: https://github.com/hku-mars/livox_camera_calib

aditdoshi333, 2022-01-04:
@ziv-lin I am using the same repo. But I think it is calibration lidar and RGB not imu and rgb. Do I have to matrix from this repo directly in the config file?

ziv-lin, 2022-01-04:
> Hello @ziv-lin ,
> 
> I think calibration is good enough because I am using the same matrix in fastlio2 (modified) for coloring point cloud and doing a decent job. But I think the way I am putting the matrix there is some with that. Is there any repo or something you can suggest for imu and rgb camera?

How about the color of the point cloud in a static environment (i.e., don't move the sensors), is that correct?

aditdoshi333, 2022-01-04:
@ziv-lin  I tried the in a static environment and it looks better but there is still some offset. The following is the output
[image]

[image]
[image]

The objects are painted on the wall in real they are on a table. Looks like there is some static offset

ziv-lin, 2022-01-04:
> @ziv-lin I am using the same repo. But I think it is calibration lidar and RGB not imu and rgb. Do I have to matrix from this repo directly in the config file?

In our work, we treat the LiDAR frame and IMU (the LiDAR built-in IMU) as the same in our implementation. By this, the extrinsic between camera and IMU is the same as the extrinsic as extrinsic of camera-LiDAR.

ziv-lin, 2022-01-04:
> @ziv-lin I tried the in a static environment and it looks better but there is still some offset. The following is the output [image]
> 
> [image]
> 
> [image]
> 
> The objects are painted on the wall in real they are on a table. Looks like there is some static offset

Is this the output of r3live or others?

aditdoshi333, 2022-01-04:
> > @ziv-lin I tried the in a static environment and it looks better but there is still some offset. The following is the output [image]
> > [image]
> > [image]
> > The objects are painted on the wall in real they are on a table. Looks like there is some static offset
> 
> Is this the output of r3live or others?

Its from r3live

aditdoshi333, 2022-01-04:
> In our work, we treat the LiDAR frame and IMU (the LiDAR built-in IMU) as the same in our implementation. By this, the extrinsic between camera and IMU is the same as the extrinsic as extrinsic of [...]

Okay great. Then I can use the matrix directly. I am attaching an image from fastlio coloring just to make sure the calibration mat is correct. 
[image]

aditdoshi333, 2022-01-04:
@ziv-lin Sorry to bother but there is one more issue. Sometimes I am getting the following error ( Same config file and same bag file)
[image]

ziv-lin, 2022-01-04:
> Its from r3live

It seems that your configuration is correct. What is the problem about R3LIVE?

aditdoshi333, 2022-01-04:
> It seems that your configuration is correct. What is the problem about R3LIVE?

The color is not correct. There is an offset in the above sample it looks as if the objects are pasted on a wall. But it is not like that. And such color is coming only if the scene is static. If I am moving the setup the color is messed up fully

ziv-lin, 2022-01-04:
Oh, sorry, my mistake... What is your image resolution?

aditdoshi333, 2022-01-04:
its 1920 x 1080

ziv-lin, 2022-01-04:
Might be I got the problem you meet with, the default image resolution I used is 1280 X 1024 and I should open this configuration for you to config.

aditdoshi333, 2022-01-04:
Okay thanks a lot @ziv-lin

ziv-lin, 2022-01-04:
Can you replace all 1280 in codes to 1920 to see is that OK? I will commit a hotfix in tonight or tomorrow.

aditdoshi333, 2022-01-04:
Okay sure, I will try and update.

ziv-lin, 2022-01-04:
https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L381

https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L387

https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L391

https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L397

https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L1069

ziv-lin, 2022-01-04:
Thank you for you reporting such issues~, this actually a bug.

aditdoshi333, 2022-01-04:
Hey @ziv-lin, I updated the image resolution. Thanks a lot for a quick fix. It improved the coloring but there is still significant bleeding. It looks like a calibration issue but I am not sure is it related to image size or aspect ratio?
 
[image]
[image]

ziv-lin, 2022-01-04:
Can you try and move the sensors to the mapping results? Our algorithm can online calibrate the extrinsic to make it more correct.

ziv-lin, 2022-01-04:
Sorry, my bad, you should also change the image_heigh 1024 to 1080, e.g., in the following codes:

https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L387
https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L397
https://github.com/hku-mars/r3live/blob/a5a4d8493ada5fed8fe90d1f1566c9abecc3ae24/r3live/src/r3live_vio.cpp#L1069

aditdoshi333, 2022-01-04:
@ziv-lin I changed height and width as you suggested. But the issue is still the same image color is bleeding near the edges. And one more thing I am still getting the following error sometimes.
[image]

Whenever I get this error I need to rerun the r3live node.

ziv-lin, 2022-01-04:
Can you share your data with me，include both your rosbag files and your configurations？

aditdoshi333, 2022-01-04:
Ya sure.

link : https://drive.google.com/drive/folders/1IPIvW-gzIYbW7z8jygCI1WhXZv1lE7YO?usp=sharing

ziv-lin, 2022-01-04:
I can not open the folder, is there any problem?

aditdoshi333, 2022-01-04:
@ziv-lin I am able to open in incognito mode also. Should I upload somewhere else then google drive?

ziv-lin, 2022-01-04:
I can download the config files, but can't get the rosbag file, can you set put them together in the same directory?

jxx315, 2022-01-04:
hi, ziv-lin,

I  share  a  short rosbag  for  calib,  with  hight  image resolution and livox avia pointcloud.    

To try r3live,   Which config file should I modify ?

  链接: https://pan.baidu.com/s/1y0xd2kICGSgKEhat-Bt3hQ  密码: kroh

 
height: 2048
width: 3072

camera matrix
1745.304795 0.000000 1519.690772
0.000000 1749.029333 1067.018842
0.000000 0.000000 1.000000

distortion
-0.102255 0.116367 -0.000031 0.003545 0.000000

extrinsic.txt
-0.016054,-0.999561,-0.0248822,-0.0664396
-0.00713346,0.0249993,-0.999662,0.0192496
0.999846,-0.0158711,-0.00753167,0.0634879
0,0,0,1

[image]
999846,-0.0158711,-0.00753167,0.0634879
0,0,0,1
[](url)

aditdoshi333, 2022-01-04:
> I can download the config files, but can't get the rosbag file, can you set put them together in the same directory?

Done. Please check and let me know

ziv-lin, 2022-01-04:
> > I can download the config files, but can't get the rosbag file, can you set put them together in the same directory?
> 
> Done. Please check and let me know

That right now, I will try your bag tonght if possible.

ziv-lin, 2022-01-04:
> hi, ziv-lin,
>
> I share a short rosbag for calib, with hight image resolution and livox avia pointcloud.
>
> To try r3live, Which config file should I modify ?
>
> 链接: [...]

Take a glance look, it seems that your result is correct?

jxx315, 2022-01-04:
> Take a glance look, it seems that your result is correct?

Sorry,  I upload  rosbag and camera param(intrinsic and extrinsic)again  in one link .  
链接: https://pan.baidu.com/s/1IMkYRubNFjIX-z5UOR5mhA  密码: deja

 I noticed that the resolution is in your work is 1280 X 1024 .    However my camera in this test is 2048 X 3072,  Which config file should I modify  in R3Live?    Change all of 1280 X 1024  to  2048 X 3072 ?

ziv-lin, 2022-01-04:
Yes, you should change all 1280 -> 2048, and 1024 -> 3072 (why is your height larger than width? is there any wrong?)
The fix of this bug will be pushed in this repo withing tease two days.

jxx315, 2022-01-04:
> Yes, you should change all 1280 -> 2048, and 1024 -> 3072 (why is your height larger than width? is there any wrong?) The fix of this bug will be pushed in this repo withing tease two days.

Sorry, my bad，Resolution is actually 2048  X 3072 (height *width) . 
Thanks very much for your reply，Looking forward to the next update

tlin442, 2022-01-05:
@ziv-lin I think the issue is that this:
https://github.com/hku-mars/r3live/blob/28f5365e4033fe1f82154a85f9f60191ca52c361/r3live/src/r3live_vio.cpp#L383-L384
Malforms the camera calibration matrix here when the camera width isn't 1280 pixels.
https://github.com/hku-mars/r3live/blob/28f5365e4033fe1f82154a85f9f60191ca52c361/r3live/src/r3live_vio.cpp#L195-L196

I had the same issue undistorting my input with a 848*800 camera.

ziv-lin, 2022-01-05:
I am now fixing and testing this problem, please wait with patience.

ziv-lin, 2022-01-05:
@aditdoshi333 It runs your data carefully, it seems that your problem is your calibration is not accurate enough, causing the color rendering is not so accurate. 
[image]
[image]

In addition, the print of ""****** Remove_outlier_using_ransac_pnp error*****"" is caused by the delayed incoming image message (see following figure). You can ignore this print since it didn't play too much effect after R3LIVE has received enough image frames. 
[image]

If your play the bag with ''-s 5'', this warning disappear:
```
rosbag play YOUR.bag -s 5
```

ziv-lin, 2022-01-05:
Hi~, @aditdoshi333 @jxx315 I have just pushed a commit that fixes this bug, now R3LIVE allows you to set the image resolution correctly, can you have a try of this version? Please let me know if you found any bugs and problems.

ziv-lin, 2022-01-05:
You can now set your own image resolution by modifying these two configs:
https://github.com/hku-mars/r3live/blob/4d386cc4a94d38cb8bc388735478c1087f9393db/config/r3live_config.yaml#L19
https://github.com/hku-mars/r3live/blob/4d386cc4a94d38cb8bc388735478c1087f9393db/config/r3live_config.yaml#L20

aditdoshi333, 2022-01-06:
@ziv-lin Thanks for the quick fix. I will run the new commit and update you in the next 2 days. And I will also check calibration at my end. Thanks for all the efforts.

One more doubt the calibration between camera and imu is also for texture mapping or it will used in mapping also. I mean preciseness of calibration is it proportional to the accuracy of mapping? 

Thank you

Camilochiang, 2022-01-06:
Hei @aditdoshi333 . In my experience a more precise calibration will give you a better making. See here for example (Both pictures are a top view):

**Not so good calibration**:
[image]
**A better calibration**:
[image]

You can clearly see that the better the calibration, the better is the alignment of the incoming scans (white points). This is of course as the intrinsic and extrinsic parameters play an important role to say from were the points are coming. I have to do a better calibration in any case, you can see than in my "better calibration" is not perfect

Question: I have to rotate the extrinsic parameters matrix when working with my data (see #21). Was your case too? I have the impression than that is not documented

jxx315, 2022-01-09:
> 

great!

Tomato1107, 2022-01-11:
> Okay great. Then I can use the matrix directly. I am attaching an image from fastlio coloring just to make sure the calibration mat is correct. [image]

Hi! @aditdoshi333 May I ask you how to add the camera rgb information to the lidar in the fast_lio, I meet some problems.

aditdoshi333, 2022-01-11:
Hello @ziv-lin ,
Sorry for the delay. I am yet to calibrate the camera and test it. I am sick so will try out asap.

aditdoshi333, 2022-01-11:
> Hi! @aditdoshi333 May I ask you how to add the camera rgb information to the lidar in the fast_lio, I meet some problems.

Hello,

You can change the struct of point cloud and add rgb. I color the incoming point cloud before mapping. I am using the livox calibration code for coloring every frame.

Tomato1107, 2022-01-11:
> Hello,
>
> You can change the struct of point cloud and add rgb. I color the incoming point cloud before mapping. I am using the livox calibration code for coloring every frame.

Thanks very much! I will try!

ziv-lin, 2022-01-11:
@aditdoshi333 I am so sorry to hear that. Please take good care of your body. Hoping every goes well.

aditdoshi333, 2022-01-12:
Hello @ziv-lin,

I can confirm that the color offset is because of poor calibration. Sorry for the trouble.
Thank you for all your efforts.

closed 2022-01-12 by aditdoshi333