hku-mars/r3live #17: 关于real-time

BG2CRW, 2022-01-04:
请问r3live做不到实时对吗，因为在跑的时候，发现越往后越慢，最终会爆内存

ziv-lin, 2022-01-04:
是RViz卡了吧？如果物理内存使用满了, 程序将会使用虚拟内存swap，在这种情况下，内存的访问速率取决你的硬盘速率。

DarrenWong, 2022-01-04:
@ziv-lin 谢谢分享好的工作，
@BG2CRW 我这边尝试了下的确会有点卡 跑hkust的数据到中后面已经吃了27个G内存，感觉长时间跑需要一个比较好的性能
[image]
[image]

ziv-lin, 2022-01-04:
这个场景是rViz卡了, 地图中的点都上几千万了，rviz肯定显示不过来，这种情况下肯定会卡的，你看odomtry的输出，应该能保证实时的odom输出

ziv-lin, 2022-01-04:
> 

你用ubuntu的系统管理器查看内存占用，
```
gnome-system-monitor 
```
这个场景下，rviz占用了9GB左右的内存，在考虑到rviz只记录了点的RGBXYZ信息，我们的r3live还记录了每个点的3X3的cov矩阵，时间戳等其他东西，所以我觉得这个内存占用应该是正常的。

DarrenWong, 2022-01-04:
谢谢林博回复, 地图大了rviz是会变卡甚至挂了，能够理解~  工作还是很棒的，再次感谢分享

ziv-lin, 2022-01-04:
是的，rviz变卡不代表我们的算法会变慢，要看实际我们的odom输出频率才行。实际上，得益于ikd-tree (https://github.com/hku-mars/ikd-Tree) 我们的计算复杂度和地图的大小的增长关系微乎其微。

ziv-lin, 2022-01-05:
In this bag (hkust_campus_00.bag), I print the total number of points in maps, find that it reaches **55334368** points (fifty-five million). Thus, I think the consumption of memory is reasonable.
[image]

OliverShaoPT, 2022-01-05:
林博您好，感谢您分享的工作。想请问，如果要把R3LIVE作为里程计使用，不保存过多的地图可以通过调整哪几个参数来实现呢？我注意到了“cube_side_length” 这个参数，但只调整这个参数并不能减少内存占用。

ziv-lin, 2022-01-05:
Tune these parameters can significantly reduce the use of memory. 
**r3live_commong/minimum_pts_size** controls the minimum point distance among points in map, increasing this value can reduce the total points in the map thereby reducing the use of memory.
https://github.com/hku-mars/r3live/blob/28f5365e4033fe1f82154a85f9f60191ca52c361/config/r3live_config.yaml#L11
In addition, reduce the working image size can also reduce the usage of RAM
https://github.com/hku-mars/r3live/blob/28f5365e4033fe1f82154a85f9f60191ca52c361/config/r3live_config.yaml#L12

ziv-lin, 2022-01-11:
This issue is closed for no more updates.
Thanks~

closed 2022-01-11 by ziv-lin