hku-mars/r3live #19: The height error in mapping

OliverShaoPT, 2022-01-04:
Hello @ziv-lin ,

Thanks for sharing your great works. The accuracy and detail of R3LIVE real-time mapping are impressive, but there are still some problems. I built R3LIVE according to the instructions and run "hkust_campus_seq_00.bag", the results I got are not consistent with the PCD file published on GitHub.
As shown in the following screenshot, the picture above is the result I got, the picture below is the PCD result published on GitHub. There exist obvious height error in my testing. In the test of FAST-LIO2, the problem of height error also appears.
Did I miss something?

[image]

[image]

Best regards

ziv-lin, 2022-01-04:
It is hard to say, in our environments, it can close the loops well by itself. The result might show very little difference due to the difference in both software and hardware environments. And,  in the sequence of HKUST_campus_00 (1.3 km) and HKUST_campuse_01 (1.5 km), the little drift (e.g., 0.001 deg) at the beginning might cause a large error at the last. 
For most of the existing slam systems, no one can guarantee that the result is the same at each time and at each computation platform.

EloyAldao, 2022-01-04:
Hi:

Where are the PCD files available to download?
I only found the bag files.

Thanks

ziv-lin, 2022-01-04:
Some of our result can be download in this repo: https://github.com/ziv-lin/r3live_preview

OliverShaoPT, 2022-01-04:
> 

You're right. My notebook has only 16GB of memory, but I noticed that your computer has more than 60 GB of memory. I'll try a better computer later. And I also want to test the upper limit of R3LIVE accuracy by adding the IMU with higher accuracy, such as Xsens 680G. (As the built-in IMU of AVIA is BMI088.)
Again, thanks your great works.

OliverShaoPT, 2022-01-05:
> 

I run the same rosbag in FAST-LIO2, it works fine, there is no obvious height error. (As shown in the following picture.)
[image]

I wonder what is the recommended hardware configuration for R3LIVE? Actually, at the last third of HKUST_campus_00 sequence, my physical memory has been used up, can we speculate that the main reason is the lack of physical memory?

[image]

ziv-lin, 2022-01-05:
It seems that you zoom with different scales, cause the drift in r3live to seem larger? In our testing environment, R3LIVE can stable closes the loop itself.

ziv-lin, 2022-01-05:
It is hard to say,  is there any loss of message due to the lack of mem?

ziv-lin, 2022-01-05:
Is R3LIVE can stable close the loop in other smaller bag?

OliverShaoPT, 2022-01-05:
> It seems that you zoom with different scales, cause the drift in r3live to seem larger? In our testing environment, R3LIVE can stable closes the loop itself.

My bad, the history point cloud in ROS_rviz will disappear over time. Here is the correct PCD save by FAST_LIO2. The height error also exists, but still better than other SLAM works. 
[image]

[image]

ziv-lin, 2022-01-05:
Both FAST-LIO2, R2LIVE/R3LIVE never expect you can close the loop by traveling up to 1.3 Km in all scenarios at all computation platforms. If you have the requirement to build a large map, implementing loop detection and closure is your choice.

Our paper put this as an example, by saying that we can close the loop itself (at least in our real word experiment, see our recorded video https://www.youtube.com/watch?v=j5fT8NE5fdg) without the aid of additional procedure (i.e., without the aid of loop closure). Aims to show that the drift in our proposed is very low, which demonstrates that our proposed method has good accuracy in localization. This does not contrary to the fact that all slam systems are drifting over time, compare to other slam systems, our proposed system drifts less.

OliverShaoPT, 2022-01-05:
> It is hard to say, is there any loss of message due to the lack of mem?

I slow down the play rate of rosbag to 0.5. The delay between rosbag and R3LIVE terminal is gone, but it couldn't close the loop itself. I'll try running it on the server later.

[image]

OliverShaoPT, 2022-01-05:
> Both FAST-LIO2, R2LIVE/R3LIVE never expect you can close the loop by traveling up to 1.3 Km in all scenarios at all computation platforms. If you have the requirement to build a large map, [...]

I totally agree with you. Compare with other SLAM works, FAST-LIO2 and R3LIVE impress me a lot. I may try to combine the FAST-LIO2 or R3LIVE with global sensors that offer absolute measurements for eliminating drift, such as GPS.

ziv-lin, 2022-01-05:
Yes, as pointed before, if you have the requirement to build a large map, implementing loop detection and closure is your choice, otherwise, fusing with other sensors for providing an absolute measurement.

closed 2022-01-05 by OliverShaoPT