hku-mars/r3live #20: rosbag replay a wired result

Marshall-Hu, 2022-01-04:
right window shows a mess pointcloud.
should I change code somewhere or miss something?
[image]

Marshall-Hu, 2022-01-05:
The console gives this kind of error:
/ws_livox/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! 0.793862  3.76128 -10.1187
here is my pc software version:
[image]

ziv-lin, 2022-01-05:
With referring to issue https://github.com/hku-mars/r3live/issues/11 
I suggest you check if the OpenCV you use for compiling the codes matches exactly the openCV in your runtime environment. That is, the OpenCV version oriented by R3LIVE should as the same as the version info printed in your terminal after types:
```
pkg-config --modversion opencv
```

Marshall-Hu, 2022-01-05:
@ziv-lin Hi, I have tried  several time to uninstall OpenCV whitch I build from source code. And I check OpenCV version:
[image]
But after delete OpenCV, I compile r3live. It success with cv 3.3.1 whitch is installed by Ros-Kinetic-visionCV :
[image]
Then I replay rosbag, i get error:
[image]

It Seems that IMU-Processing works in the wrong way. Should I remove kinetic-ros opencv, and then try to build cv from source code?

Marshall-Hu, 2022-01-05:
I also try build cv 3.3.1 from source code. and then `pkg-config --modversion opencv`check OpenCV version with 3.3.1.
Run and get IMU_Processing errors. The result is that Lidar moving Path and Color Point Cloud not mapping as expected.

Marshall-Hu, 2022-01-06:
I change `set(OpenCV_DIR "/home/cmown/opencv-3.4.16/build")` in the r3live CMakeLists.txt . And console outputs like this:
[image]
But when it success compiling, run and get opencv3.3.1:
[image]
type `pkg-config --modversion opencv`, I got opencv3.4.16. So  I really don't know how to set opencv right version

Camilochiang, 2022-01-06:
So it look that you have a mess.
1. First starting by removing all your custom installations (you may need to install purge)
```sudo apt-get purge '*opencv*' ```  
now check the version of opencv
```pkg-config --modversion opencv```

2. Re instally opencv
```sudo apt-get install build-essential
sudo apt-get install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev
# I download the 3.4.16 version and decompress on /home/ubuntu
cd /home/ubuntu/opencv-3.4.16
mkdir build
cd build
cmake -D CMAKE_BUILD_TYPE=Release -D CMAKE_INSTALL_PREFIX=/usr/local ..
sudo make install
pkg-config --modversion opencv```

ziv-lin, 2022-01-06:
Make sure that the LIBRARY you build and run with the same is **very very important** to achieve the correct result you expected. If you have some troubles in modifying your current software environment, I suggest your try R3LIVE in a new clean machine to see if it work.

Marshall-Hu, 2022-01-06:
> 
Hi C.
thanks to giving these steps. After doing this and `catkin build` r3live, it cause cmake error: Missing cv-brige . So I have to apt-get install this ros-xxx-cv-bridge. But this apt instruction lead to install ROS-Opencv with 3.3.1 in the path "/opt/ros/kinetic/include/opencv-3.3.1-dev". Finally r3live compliled with 3.3.1 not 3.4.16
My confuseing is why cv3.3.1 works in a wired way in my first post image. I notice that I installed ` Sophus ` lib with fuzzy CV version, does this cause some trouble?

Marshall-Hu, 2022-01-06:
> Make sure that the LIBRARY you build and run with the same is **very very important** to achieve the correct result you expected. If you have some troubles in modifying your current software environment, I suggest your try R3LIVE in a new clean machine to see if it work.

> So it look that you have a mess.
> 
>     1. First starting by removing all your custom installations (you may need to install purge)
>        `sudo apt-get purge '*opencv*' `
>        now check the version of opencv
>        `pkg-config --modversion opencv`
>        This should give you the default version in your machine before all your installations or none, not sure.
> 
>     2. Re instally opencv
> 
> 
> ```
> sudo apt-get install cmake git libgtk2.0-dev pkg-config libavcodec-dev libavformat-dev libswscale-dev
> # I download the 3.4.16 version and decompress on /home/ubuntu
> cd /home/ubuntu/opencv-3.4.16
> mkdir build
> cd build
> cmake -D CMAKE_BUILD_TYPE=Release -D CMAKE_INSTALL_PREFIX=/usr/local ..
> sudo make install
> pkg-config --modversion opencv
> ```
> 
> ps: dont change anything of r3live dude. your problem is on opencv, not r3live

Hi C.
thanks to giving these steps. After doing this and `catkin build` r3live, it cause cmake error: Missing cv-brige . So I have to apt-get install this ros-xxx-cv-bridge. But this apt instruction lead to install ROS-Opencv with 3.3.1 in the path "/opt/ros/kinetic/include/opencv-3.3.1-dev". Finally r3live compliled with 3.3.1 not 3.4.16
My confuseing is why cv3.3.1 works in a wired way in my first post image. I notice that I installed ` Sophus ` lib with fuzzy CV version, does this cause some trouble?

Marshall-Hu, 2022-01-06:
> Make sure that the LIBRARY you build and run with the same is **very very important** to achieve the correct result you expected. If you have some troubles in modifying your current software environment, I suggest your try R3LIVE in a new clean machine to see if it work.

Sure that is final way. Thanks to your nice work Lin! Cheer For U! 
ps:any new i will post here :d

Camilochiang, 2022-01-06:
What do you get now from `pkg-config --modversion opencv` ?

Camilochiang, 2022-01-06:
Maybe repeat the process installing cv-bridge before..

Marshall-Hu, 2022-01-06:
> What do you get now from `pkg-config --modversion opencv` ?

if i purge opencv*, then i got 3.4.16(because i build it myself and make install. and i `make uninstall`, then i got nothing. it say missing opencv.

Camilochiang, 2022-01-06:
If you install opencv 3.4.16 and install from source cv-bridge, then you should be fine... 
https://github.com/ros-perception/vision_opencv

Marshall-Hu, 2022-01-06:
> If you install opencv 3.4.16 and install from source cv-bridge, then you should be fine... https://github.com/ros-perception/vision_opencv

Yeah, I finally get the right answer.
And through my trip, Eigen version and opencv version should be kind of couple:
cv3.4.16 seems not working with Eigen.version > 3.3.7 while compile r3live. so here is my:
[image]

Thx again C.

closed 2022-01-06 by Marshall-Hu

mameesie, 2022-05-29:
> @ziv-lin Hi, I have tried several time to uninstall OpenCV whitch I build from source code. And I check OpenCV version: [image] But after delete OpenCV, I compile r3live. It success with cv 3.3.1 [...]

Hi, I still have this error even after building and running with same openCV. The error occurs when I capture objects at < 2 meters. when capturing objects at long range it will work perfectly. Is this normal behaviour? 

[image]

cywm39, 2023-09-15:
@mameesie I have the same problem. Have you solved it?

mameesie, 2024-05-22:
@cywm39 Sorry for the late reply I didnt use github for a while, no i did not solve it, did you?