hku-mars/r3live #22: question about lic_point_cloud_undistort

h-k8888, 2022-01-04:
https://github.com/hku-mars/r3live/blob/28f5365e4033fe1f82154a85f9f60191ca52c361/r3live/src/loam/IMU_Processing.cpp#L415
the way of point cloud undistort is different with fastlio2, can you please explain why?
**state_inout.rot_end.transpose()** is the rotation from Global to IMU at timestamp e, 
**R_i * Lidar_offset_to_IMU** is the compensation for lidar translation by angular velocity in Global frame,
but **pos_liD_e** is lidar pose at timestamp e in global frame,  why not "- state_inout.pos_end"?

ziv-lin, 2022-01-05:
The LIO-subsystem in R3LIVE/R2LIVE diverged with FAST-LIO2 at a very early time in development. I think the point cloud undistortion module among these works is OK in compensation for the in-frame LiDAR distortion, at least I can not seek out any bugs and errors in these codes.

h-k8888, 2022-01-05:
> The LIO-subsystem in R3LIVE/R2LIVE diverged with FAST-LIO2 at a very early time in development. I think the point cloud undistortion module among these works is OK in compensation for the in-frame LiDAR distortion, at least I can not seek out any bugs and errors in these codes.

Thanks for your reply.
If there are large external parameters between IMU and lidar, I think this is not accurate enough. And the rotation matrix of external parameters is not used in distortion correction to convert the point cloud to IMU?

ziv-lin, 2022-01-05:
Yes, from this point. You are right. I will consider modifying this code in the coming future. Thank you for your corrections.

h-k8888, 2022-01-07:
> Yes, from this point. You are right. I will consider modifying this code in the coming future. Thank you for your corrections.
Some functions in r3live_lio.cpp, like  
 https://github.com/hku-mars/r3live/blob/4d386cc4a94d38cb8bc388735478c1087f9393db/r3live/src/r3live_lio.cpp#L200
the rotation matrix of LiDAR and IMU is not used, which may cause some problems.This is probably the reason for the poor matching between camera and LiDAR in other issues, especially when the equipment has large external parameters.

linyayunyayun, 2022-01-11:
I'm also confused why you didn't use Lidar_R_to_IMU, or just make it an identity matrix.
If I add Lidar_R_to_IMU in distortion compensation, will there be other inconsistency with the overall code, except for those projection functions mentioned above?

ziv-lin, 2022-01-11:
Copied, this part of the code will be optimized and tested in the future, I add this item to my TODO list now.

closed 2022-02-08 by stale[bot]

seajayshore, 2022-03-17:
Hello, I'm curious - is this issue still unresolved?
If I know the Lidar-to-IMU extrinsic would it be helpful to transform the incoming point cloud data to IMU frame straight away when new messages arrive? Maybe easier than fixing things mentioned above deeper in the code?

iorileslie, 2025-03-10:
> Copied, this part of the code will be optimized and tested in the future, I add this item to my TODO list now.复制，这部分代码将在未来进行优化和测试，我现在将此项目添加到我的TODO列表中。

is there any result?