hku-mars/r3live #27: REQUIRED process [r3live_mapping-3] has died!

kxch928925819, 2022-01-06:
REQUIRED process [r3live_mapping-3] has died!
process has died [pid 4660, exit code -11, cmd /home/dji/catkin_ws/devel/lib/r3live/r3live_mapping __name:=r3live_mapping __log:=/home/dji/.ros/log/94ab4d12-6ea4-11ec-b07f-1c697aacadba/r3live_mapping-3.log].
log file: /home/dji/.ros/log/94ab4d12-6ea4-11ec-b07f-1c697aacadba/r3live_mapping-3*.log
Initiating shutdown!

closed 2022-01-11 by ziv-lin

BoyLazyA, 2022-01-18:
I have encountered the same bug, have you solved it?

treememory, 2022-02-09:
I have encountered the same bug too！
in ~/.ros/log/46ca3042-89aa-11ec-83ff-4851c5f254c2/master.log

> [rosmaster.master][INFO] 2022-02-09 21:15:05,182: -PUB [/laser_cloud_sharp] /r3live_LiDAR_front_end http://ll:44233/
> [rosmaster.master][INFO] 2022-02-09 21:15:05,183: -SUB [/livox/lidar] /r3live_LiDAR_front_end http://ll:44233/
> [rosmaster.master][INFO] 2022-02-09 21:15:05,183: -SERVICE [/r3live_LiDAR_front_end/get_loggers] /r3live_LiDAR_front_end rosrpc://ll:44559
> [rosmaster.master][INFO] 2022-02-09 21:15:05,183: -SERVICE [/r3live_LiDAR_front_end/set_logger_level] /r3live_LiDAR_front_end rosrpc://ll:44559
> [rosmaster.master][INFO] 2022-02-09 21:15:05,280: publisherUpdate[/rosout] -> http://ll:46605/ ['http://ll:43601/', 'http://ll:38067/', 'http://ll:44369/']
> [rosmaster.master][INFO] 2022-02-09 21:15:05,281: publisherUpdate[/rosout] -> http://ll:46605/ ['http://ll:43601/', 'http://ll:38067/', 'http://ll:44369/']: sec=0.00, result=[1, '', 0]
> [rosmaster.master][INFO] 2022-02-09 21:15:05,281: publisherUpdate[/laser_cloud_flat] -> http://ll:38067/ []
> [rosmaster.master][INFO] 2022-02-09 21:15:05,281: publisherUpdate[/laser_cloud_flat] -> http://ll:38067/ []: sec=0.00, exception=[Errno 111] Connection refused
> [rosmaster.threadpool][ERROR] 2022-02-09 21:15:05,282: Traceback (most recent call last):
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/threadpool.py", line 218, in run
>     result = cmd(*args)
>   File "/opt/ros/melodic/lib/python2.7/dist-packages/rosmaster/master_api.py", line 210, in publisher_update_task
>     ret = xmlrpcapi(api).publisherUpdate('/master', topic, pub_uris)
>   File "/usr/lib/python2.7/xmlrpclib.py", line 1243, in __call__
>     return self.__send(self.__name, args)
>   File "/usr/lib/python2.7/xmlrpclib.py", line 1602, in __request
>     verbose=self.__verbose
>   File "/usr/lib/python2.7/xmlrpclib.py", line 1283, in request
>     return self.single_request(host, handler, request_body, verbose)
>   File "/usr/lib/python2.7/xmlrpclib.py", line 1311, in single_request
>     self.send_content(h, request_body)
>   File "/usr/lib/python2.7/xmlrpclib.py", line 1459, in send_content
>     connection.endheaders(request_body)
>   File "/usr/lib/python2.7/httplib.py", line 1095, in endheaders
>     self._send_output(message_body)
>   File "/usr/lib/python2.7/httplib.py", line 898, in _send_output
>     self.send(msg)
>   File "/usr/lib/python2.7/httplib.py", line 860, in send
>     self.connect()
>   File "/usr/lib/python2.7/httplib.py", line 837, in connect
>     self.timeout, self.source_address)
>   File "/usr/lib/python2.7/socket.py", line 575, in create_connection
>     raise err
> error: [Errno 111] Connection refused

found a solution: https://github.com/Livox-SDK/livox_mapping/issues/1
but, i want in ubuntu18.04,  now have no idea.....

Camilochiang, 2023-03-21:
I encounter this problem after coming back to this algorithm to check something
- Basically the packages are been refused as either your firewall is blocking them or your ports are not open
- Additionaly I encounter the fact that docker (not sure why?) create a higher priority routing destination. I modifed using `ifmetric`. After reducing the priority with ```` sudo ifmetric docker0 200```` it even dissapear (weird...)

L-5022, 2023-03-23:
I have faced the same problem and resolved it through [您好，请问您在测试的时候有遇到过这个问题吗 #11](https://github.com/hku-mars/r3live/issues/11)。
I think it is about opencv. When installing ros, opencv3.2 has been automatically installed, but I also installed opencv4.5.3 myself, 
so I modified lines 94, 96, and 119 in (/opt/ros/melodic/share/cv_bridge/make/cv_bridgeConfig.cmake) to correspond to 4.5.3