hku-mars/r3live #29: Documentation: topic details

Camilochiang, 2022-01-06:
Hei. I'm trying to get some information from some topics. Could you please confirm the following:
1. /path :  Correspond to the IMU path at the time of recording using the LIDAR time
2. /camera_path: Correspond to the camera path at the time of recording using the camera time
3 . /camera_odom : same than camera_path but with running time

Also, at specific times I would like to publish all points that are visible in the camera field of view, or just simply for now all the points. Do you know which topic I should use for that? there are multiple RGB_map topics, so one option will be to hijack your cv_keyboard_callback and trigger it from another topic/sign. What do you think?

ziv-lin, 2022-01-07:
I will try to add documentation if I have plenty of time in the future...  I have a busy schedule recently, and I have no idea when I will be free [cry][cry][cry].

ziv-lin, 2022-01-11:
This issue is closed for no more updates.
Thanks~

closed 2022-01-11 by ziv-lin