hku-mars/r3live #30: Fixme: Catching a cv exception, DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences.

qiaozhijian, 2022-01-07:
Hi, Lin. R3live is really an amazing job. 

I ran into a bug when testing "hkust_campus_seq_00". Do you have any ideas? It seems to appear around the 15881st image and the 10587th LiDAR.

| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   23:38:24     |    10587        |    15881           | 54814318     |    26043          
Catching a cv exception: OpenCV(3.4.10) /home/qzj/Downloads/tool/opencv-3.4.10/modules/calib3d/src/calibration.cpp:1099: error: (-2:Unspecified error) in function 'void cvFindExtrinsicCameraParams2(const CvMat*, const CvMat*, const CvMat*, const CvMat*, CvMat*, CvMat*, int)'
> DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. (expected: 'count >= 6'), where
>     'count' is 5
> must be greater than or equal to
>     '6' is 6

My environment:

[image]

ziv-lin, 2022-01-07:
Yes, I have met some strange errors with some versions of OpenCV too.  When I switch to another version, such exceptions disappear immediately...

For this exception, I use try {...} catch {...} to handle this case, shown as below:
https://github.com/hku-mars/r3live/blob/4d386cc4a94d38cb8bc388735478c1087f9393db/r3live/src/rgb_map/rgbmap_tracker.cpp#L343-L353

And, since it has very little in affecting the overall system (e.g., crash the whole program, cause the large drift in localization), I temporally skip this exception by ignoring this image frame.
However, I will try to fix it if I come up with a better idea to handle this problem.

ziv-lin, 2022-01-07:
I know the DLT algorithm needs at least 6 points for pose estimation from 3D-2D point correspondences. But such an OpenCV exception should not happen since I have guaranteed that the minimal 3D-2D much larger than 10, see:

https://github.com/hku-mars/r3live/blob/4d386cc4a94d38cb8bc388735478c1087f9393db/r3live/src/rgb_map/rgbmap_tracker.cpp#L336-L339

So, I have no idea that why such an exception can appear. Might be there have some bugs in function cv::solvePnPRansac of 
 some version of OpenCV (I guess)?

closed 2022-01-08 by qiaozhijian