hku-mars/r3live #31: Reconstruct and texture  mesh failed!

Leonard-Yao, 2022-01-07:
Hi, Lin. It's really a great job, especially the reconstruct module.
However, I occurred some weird problems.

> =============================================================
App name   : R3LIVE: A Robust, Real-time, RGB-colored, LiDAR-Inertial-Visual tightly-coupled state Estimation and mapping package
Build date : Jan  7 2022  14:23:18
CPU infos  : 11th Gen Intel(R) Core(TM) i7-11800H @ 2.30GHz
RAM infos  : **31.21GB** Physical Memory 24.81GB Virtual Memory
OS  infos  : Linux 5.4.0-92-generic (x86_64)
=============================================================
+++++++++++++++++++++++++++++++++++++++++++++++
Here is the your software environments: 
GCC version          : 8.4.0
Boost version        : 1.65.1
Eigen version        : 3.3.4
OpenCV version       : **3.4.16**
+++++++++++++++++++++++++++++++++++++++++++++++

I can't reconstruct and texture mesh when I use hkust_campus_seq_01.bag .And sometimes with the error 
`catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -8.28157 -10.0624 -6.91589`

It seems that there is only  RGB_map_0 point cloud.
[image]

After save the map , the point cloud is very sparse.

[image]

so the final illustration as follows(include rgb_pt.pcd):
[image]

ziv-lin, 2022-01-07:
When appear the print of 
```
catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -8.28157 -10.0624 -6.91589
```
This means your shouldn't setup up your environment well, I suggest you see these two issues here: 
https://github.com/hku-mars/r3live/issues/20
https://github.com/hku-mars/r3live/issues/11
Hope you can find your solutions soon.

Leonard-Yao, 2022-01-11:
> When appear the print of

> catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -8.28157 -10.0624 -6.91589

> This means your shouldn't setup up your environment well, I suggest you see these two issues here:

> #20

> #11

> Hope you can find your solutions soon.

Thanks for your reply. I have tried several strategy, and I'm not sure it's blamed to opencv.

[image]

when  I built it， I got warning :
> warning: enumeral mismatch in conditional expression: ‘Eigen::StorageOptions’ vs ‘Eigen::Default_t’ [-Wenum-compare]

Then play hkust_campus_seq_01.bag. Before moving device, the system was not stable with the error:
 

> catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!!

After moving, it seems that there was the first frame of the RGB map :
[image]

but registered point cloud map can be built very well.
[image]

Whether Eigen also lead to such error:

> catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!!

ziv-lin, 2022-01-11:
Is other bags exist the same error?

Leonard-Yao, 2022-01-11:
@ziv-lin 
[image]

> Is other bags exist the same error?

hkust_campus bags can run, others  run with the error :

> catkin_r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!!

[image]

ziv-lin, 2022-01-11:
Your mapping is so wired...
What is the output of your terminal after you type:
```
pkg-config --modversion opencv
```

Leonard-Yao, 2022-01-11:
@ziv-lin **version: 3.4.16** . I have uninstall all the other opencv version(include ros) . The eigen is also unique . opencv with cuda .

ziv-lin, 2022-01-11:
Hmm, I have no clue why your result is so weird... 
Since there are a  large number of users had successfully reproduced our result (see the following links), I guess (I have no evidence) that there might be some strange configurations in your environment that cause the problem so weird.

Lastly, if possible, can you try it in a totally new environment or machine? 

---------------------------------------------------
Reference:
https://blog.csdn.net/handily_1/article/details/122271243
https://zhuanlan.zhihu.com/p/453038352
https://www.bilibili.com/video/BV1eb4y1e7Ui?from=search&seid=12262955863144862401&spm_id_from=333.337.0.0

Leonard-Yao, 2022-01-12:
@ziv-lin Thank U, I change the operation version to ubuntu 20.04 and rebuilt the project. It seems that it works very well and can reconstruct successfully. And it's no need to consider the problem of opencv version . By the way, I notice that you use boost library in this work, whether you will consider to develop multi-atlas function like orb-slam3, which is very useful in AR/VR?

ziv-lin, 2022-01-12:
Nice, happy to hear that you have solved your problem. Even though I have no idea why your result was so weird on your previous computer. 

I have run orb-slam3 previously in this early year, which is actually, really great work in this field of research. However, I haven't had a careful look at its implementation yet. And, if possible, I will consider adding the multi-atlas functions by referring to orb-slam3.

Leonard-Yao, 2022-01-17:
@ziv-lin Hi, I can reconstructed mesh textured_mesh.ply, but I have no idea how to import *.ply file into UE4 . It seems that UE4 dosen't support to import this kind of file directly on ubuntu.  Can you tell me your method to achieve it. Thanks advance.

closed 2022-02-08 by stale[bot]