hku-mars/r3live #32: Build fail on Arm

chenosaurus, 2022-01-07:
Build is crashing due to dependency to `cpuid.h`, which doesn't exist on Arm.

```
In file included from /home/dchen/catkin_ws/src/r3live/r3live/src/tools/test_timer.cpp:49:0:
/home/dchen/catkin_ws/src/r3live/r3live/src/tools/tools_logger.hpp:1323:10: fatal error: cpuid.h: No such file or directory
 #include <cpuid.h>
          ^~~~~~~~~
compilation terminated.
r3live/r3live/CMakeFiles/test_timer.dir/build.make:62: recipe for target 'r3live/r3live/CMakeFiles/test_timer.dir/src/tools/test_timer.cpp.o' failed
make[2]: *** [r3live/r3live/CMakeFiles/test_timer.dir/src/tools/test_timer.cpp.o] Error 1
CMakeFiles/Makefile2:4821: recipe for target 'r3live/r3live/CMakeFiles/test_timer.dir/all' failed
make[1]: *** [r3live/r3live/CMakeFiles/test_timer.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
In file included from /home/dchen/catkin_ws/src/r3live/r3live/src/loam/LiDAR_front_end.cpp:5:0:
/home/dchen/catkin_ws/src/r3live/r3live/src/loam/../tools/tools_logger.hpp:1323:10: fatal error: cpuid.h: No such file or directory
 #include <cpuid.h>
          ^~~~~~~~~
compilation terminated.
r3live/r3live/CMakeFiles/r3live_LiDAR_front_end.dir/build.make:62: recipe for target 'r3live/r3live/CMakeFiles/r3live_LiDAR_front_end.dir/src/loam/LiDAR_front_end.cpp.o' failed
make[2]: *** [r3live/r3live/CMakeFiles/r3live_LiDAR_front_end.dir/src/loam/LiDAR_front_end.cpp.o] Error 1
CMakeFiles/Makefile2:4954: recipe for target 'r3live/r3live/CMakeFiles/r3live_LiDAR_front_end.dir/all' failed
make[1]: *** [r3live/r3live/CMakeFiles/r3live_LiDAR_front_end.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j2 -l2" failed
```

ziv-lin, 2022-01-08:
R3LIVE currently can not be compiled and run on the ARM platform. We will try to make it capable to it in future if possible.

SYSUHPK, 2022-01-09:
well, me too. I try to build it on tx2, all dependencies have been installed, but it reports error when compiling r3live, is it the same problem because of arm?:
[image]

SYSUHPK, 2022-01-09:
> R3LIVE currently can not be compiled and run on the ARM platform. We will try to make it capable to it in future if possible.
well, me too. I try to build it on tx2, all dependencies have been installed, but it reports error when compiling r3live, is it the same problem because of arm?:
[image]

ziv-lin, 2022-01-09:
Yes, R3LIVE currently only supports running on platforms with the SSE instruction set.

closed 2022-01-10 by chenosaurus

jonyktan, 2022-02-17:
Hi just wanted to share that by applying the fix from [here](https://github.com/ssj71/rkrlv2/commit/540dc22a912a791e0dbb493a63d8fc6f1dcbb9a3) to line 6 of "r3live/CMakeLists.txt", the `msse` error is overcome. 

However, `cpuid.h` error persists as it is referenced in "r3live/src/tools/tools_logger.hpp".
[image]

Hoping arm support can be up soon so that we can try running this on arm edge computers!

Camilochiang, 2022-02-23:
> Yes, R3LIVE currently only supports running on platforms with the SSE instruction set.

@ziv-lin Is this a priority in your pipeline? How long do you think that will it take you?  just curious. Thanks!

sumedhfb, 2024-03-27:
@ziv-lin @Camilochiang @jonyktan @SYSUHPK I am looking at building this repository for an edge-device as well. Is there an update with making it compatible with ARM64?

Camilochiang, 2024-03-28:
> @ziv-lin @Camilochiang @jonyktan @SYSUHPK I am looking at building this repository for an edge-device as well. Is there an update with making it compatible with ARM64?

No. I did change to FAST-LIVO2 due that exact reason.

sumedhfb, 2024-03-29:
> > @ziv-lin @Camilochiang @jonyktan @SYSUHPK I am looking at building this repository for an edge-device as well. Is there an update with making it compatible with ARM64?
> 
> No. I did change to FAST-LIVO due that exact reason.

That worked for me too, thank you