hku-mars/r3live #34: Tuning R3live for more robustness

aditdoshi333, 2022-01-11:
Hello @ziv-lin,

I am using r3live and I just want to thank you for such amazing work. R3live is quite robust compare to other SLAM work. I just want to know is it possible to tune r3live for more robustness? I have enough computing capacity and I don't need real-time also. 

Is it possible to change a few params in the config file by which r3live can work better in low feature areas? And is there any effect of lighting conditions on mapping performance? I understand that texture would be dark but other than that any effect on the map generation part? 

Thank you

ziv-lin, 2022-01-11:
From my view, tuning the parameter have a very limited effect on improving the robustness of R3LIVE, the default parameters work well enough in most cases. 

Actually, we are still working to make R3LIVE more robust. And, as you know, R3LIVE is of our work at 3 months ago, and we are in developing the next version of R3LIVE. 

For the future of our work, please subscribe to the update of our publications and GitHub of our labs.

Reference:
https://mars.hku.hk/
https://github.com/hku-mars
https://scholar.google.com/citations?user=V-eYCF8AAAAJ&hl=zh-CN
https://scholar.google.com/citations?user=JUHU33cAAAAJ&hl=zh-CN

aditdoshi333, 2022-01-12:
Hello @ziv-lin,

Yeah, I understand that. Default params are working really well in most of the cases. 

I just have a doubt is there any effect of lighting conditions and poor calibration on mapping performance? I understand that texture would be dark but other than that any effect on the map generation part?

Thank you

ziv-lin, 2022-01-12:
I think it is hard for me to analyze the performance of R3LIVE in those low-light and poor calibration environments.  And generally speaking, our VIO subsystem is preferred in those static (i.e., have no dynamic object) environments with enough visual texture (i.e., high contrast). 

And, if the environment can not be changed in your case (e.g., add more lights ), I suggest you use a high dynamic range camera with a larger FoV. More over, If you are not care about the sampling time, I suggest you increase the exposure time and lower your moving speed in order to sample a better image.

closed 2022-02-08 by stale[bot]