hku-mars/r3live #35: 如何录制自己的数据集？

ailiwei17, 2022-01-12:
您好，我是一个刚接触SLAM的新手，在运行完您的数据集后，我也想录制一个自己的数据集，我使用了livox mid-40和zed-2i录制，并标定了相机和雷达的外参，修改了roslaunch文件的参数
```
    <param name="LiDAR_pointcloud_topic" type="string" value= "laser_cloud_flat" />
    <param name="IMU_topic" type="string" value= "/zed2i/zed_node/imu" />
    <param name="Image_topic" type="string" value= "/zed2i/zed_node/left/image_rect_color" />
```
但rviz打开后没有任何输出，我的理解是即使imu没有进行标定，也应该会获得一个不太好的结果，请问是哪里出现了问题吗
[image]

ziv-lin, 2022-01-12:
1. Have you configured the VIO parameters correctly?
2. Can the timestamp among these messages be recorded in the same time-base (e.g., the OS system time)
3. How it was possible for a bag file with only 64.7Mb that includes 1508 images? Is there anything wrong?

ailiwei17, 2022-01-13:
感谢您的回答！
我已经修改了`image_width`，`image_height`，`camera_intrinsic`，`camera_dist_coeffs`,`camera_ext_R`,`camera_ext_t`等参数，另外重新录制了图像(之前的图像节点确实有问题)，也在rosbag后对齐时间戳，可是貌似所有节点都无法正常启动，或许是我还应该修改一些别的地方吗？
[image]

cly20-cqu, 2022-01-17:
这个问题和你的差不多，你可以看一下 https://github.com/hku-mars/r2live/issues/56

ailiwei17, 2022-01-17:
好的谢谢

closed 2022-01-17 by ailiwei17

cly20-cqu, 2022-02-16:
> 好的谢谢

> 您好，我是一个刚接触SLAM的新手，在运行完您的数据集后，我也想录制一个自己的数据集，我使用了livox mid-40和zed-2i录制，并标定了相机和雷达的外参，修改了roslaunch文件的参数
>
> ```
>     <param name="LiDAR_pointcloud_topic" type="string" value= "laser_cloud_flat" /> [...]
请问你是用的什么算法进行相机和雷达的外参标定？