hku-mars/r3live #36: Holes in reconstructed mesh

aditdoshi333, 2022-01-13:
Hello @ziv-lin,

I tried r3live to build large-area meshes. In most cases, it is doing a pretty good job but in some, I got the following problem. 
[image]
[image]
[image]

I am getting such holes all over the surface. It is quite strange as the point is uniform and there are no holes. I am running the default config and launch file. Let me know if I can improve this by any means. 

Thanks

aditdoshi333, 2022-02-07:
Hello @ziv-lin,

Can you suggest something for this issue?

ziv-lin, 2022-02-08:
Try to increase the value of close_holes_dist to close such holes?
https://github.com/hku-mars/r3live/blob/b5855c817e7b00b5b933393bfabae71420ca8a00/r3live/launch/r3live_reconstruct_mesh.launch#L29

closed 2022-02-12 by ziv-lin