hku-mars/r3live #41: LiDAR / camera frames and points captured but no RGB points

VisionaryMind, 2022-01-17:
Everything has been setup properly on our end. We are using a Livox Avia and RealSense D455 RGB stream. Your custtom Livox driver has been implemented. Unfortunately, frames are captured and points added to the map, but no RGB points are displayed in the RViz panel. 

```
| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   18:22:26  |    691      |    1000      | 1647769      |    994          I capture the keyboard input!!!
```
Further, when attempting to save the map, the following is output in the terminal:

```
Export the whole offline map to file: /home/vmind/r3live_output/test.r3live
Saving globale map...
Saving global map: 100 %
Save globale map cost 270.114 ms
Saving views and camera poses 100 % ...
Save views and camera poses cost 72.0792 ms
Save Rgb points to /home/visionarymind/r3live_output/rgb_pt
Saving offline map 100% ...
Total have 0 points.
Now write to: /home/vmind/r3live_output/rgb_pt
```

The r3live_bag.launch file has been modified accordingly to accept the RealSense image stream:

`    <param name="Image_topic" type="string" value= "/camera/color/image_raw" />`

Would you have any advice on what issue could be or how to troubleshoot further?

VisionaryMind, 2022-01-17:
We been testing this all day now, with various calibrations and settings, and it would appear the R3LIVE code, as is, simply isn't compatible with RealSense-type cameras. It has no issue utilizing LiDAR data from a Livox Avia, but regardless of how accurate the intrinsics / extrinsics are, the /camera/color/image_raw topic from a RealSense D455 is shown in the R3LIVE Rviz view, but not integrated with the map. It's impossible to say why without knowing the code more deeply. Perhaps at some point,  your team will have an opportunity to test with other cameras. For now, I can say that this project is incompatible with:

- All USB webcams (using the ROS usb_cam driver)
- RealSense D455
- Azure Kinect

Perhaps it is that the frame rate on these devices (15 / 30 FPS) is too slow for your syncing algorithm.

Camilochiang, 2022-01-17:
Your conclusions  are way too fast.  I’m using this project and the previous ones of this lab,  without problem using a real sense camera.
Are you sure that you configured properly all the topics ?

Tomato1107, 2022-01-18:
> 

Hi @VisionaryMind, I met the same problem as you before, I commented out my own camera_ext_R, camera_ext_T,  and changed them into 
    camera_ext_R:
       [0, 0, 1,
       -1, 0, 0,
       0, -1, 0]
   camera_ext_t: [0,0,0] 
Then the pointclouds had the RGB colors. I do not know why.
The usb camera that I used is 1920X1080, 30 fps, but I still have to deal with the sync, though I can run r3live, but it is very unstable..

VisionaryMind, 2022-01-18:
@Tomato1107, you are absolutely right. If I change only the camera_ext_R to the default values, R3LIVE produces an RGB cloud, but the cloud, of course, is not aligned. Time sync is very stable if a BAG file is first recorded then played back. What are you using to generate an extrinsic matrix for your camera and sensor? I still believe that is where the problem lies, and it could be that the extrinsic created automatically by tools such as [livox_camera_calibration](https://github.com/hku-mars/livox_camera_calib) is not compatible, as is, with R3LIVE. 

A closer look shows that the extrinsic R3LIVE is using in the sample appears to be transposed. Here is the R3LIVE sample extrinsic:

```
   camera_ext_R:
         [-0.00113207, -0.0158688, **0.999873**,
            **-0.9999999**,  -0.000486594, -0.00113994,
            0.000504622,  **-0.999874**,  -0.0158682]
   # camera_ext_t: [0.050166, 0.0474116, -0.0312415] 
   camera_ext_t: [0,0,0] 
```

And here is the extrinsic we have been using, directly copied from the hku-mars calibration tool output:

   camera_ext_R:
        [-0.0239835,**-0.999558**,-0.0175859
         -0.000931554,0.0176133,**-0.999844**,
          **0.999712**,-0.0239634,-0.00135357]
   camera_ext_t: [-0.0137936,0.0110846,0.108084] 

Do you see what I am saying? Values [0,1] and [1,2] should be close to -1 and [2,0] should be close to 1 as they are in the R3LIVE extrinsic above. The hku-mars calibration tool outputs them transposed, as shown above. Transposing the matrix provided by the hku-mars tool appears to fix this problem.

@Camilochiang, which calibration tool did you use to derive the values shown in the sample? Perhaps this will help us understand in which format R3LIVE requires the matrix. For now, changing transposing our matrix as follows fixes allows the RGB points to be generated with the proper calibration:

```
   camera_ext_R:
        [-0.0239835,-0.000931554,0.999712,
        -0.999558,0.0176133,-0.0239634,
        -0.0175859,-0.999844,-0.00135357]
```

Camilochiang, 2022-01-18:
@VisionaryMind . that is why I say that your conclusions were too fast. Did you look before all the issues/questions? 

Im a normal user of this tool , but a quick review of the previous issues will show you that you are not the first one to encounter these ones, which are trigger due two reasons: 
1. As you mention a rotated transformation matrix, that can be easily solve rotating the matrix and 
2. has @Tomato1107 demonstrate, a non well calibrated transformation matrix, what make the algorithm unstable

Please take the time to read #16 and #21. Good luck!

Tomato1107, 2022-01-18:
@Camilochiang For my current cituration, I can run r3live for a little while, I am trying to use other cameras, thanks!

closed 2022-01-19 by VisionaryMind