hku-mars/r3live #42: 还是运行测试的时候的那个问题

BoyLazyA, 2022-01-18:
在保证运行opencv和编译opencv版本都是3.4.16后，还是一直会崩掉，运行内存时16G.
[image]
[image]
[image]
还有就是，rqt_image_view命令后也是能看见图片的，这个怎么解决呀？

ChunqiuYang, 2022-01-20:
I have the same problem. How can we solve it?

BoyLazyA, 2022-01-21:
> 

sorry, I do not solve it,

ToutDonner, 2022-01-22:
可以尝试一下拆卸系统自带的opencv3.2.0（我的是melodic），然后将最新版的cvbridge和r3live放在同一个src下进行编译。我的电脑在编译livox_camera_calib时会出现这个错误，我按照上述方法解决的。

BoyLazyA, 2022-01-24:
> 

照做还是不行，很奇怪。

Divenire1998, 2022-01-25:
我将Boost的切换到1.65.1后，运行正常了，楼主试试呢。 但是我的Rviz显示还有点问题，不过至少能跑了 hh

closed 2022-02-09 by stale[bot]