hku-mars/r3live #48: [Question] Is there communication from VIO to LIO?

Huguet57, 2022-02-06:
Hi,
I have re-read the code and paper and I'm there's **no** communication from VIO to LIO. Correct me if I'm wrong.

What is this preferable than R2LIVE that was a factor-graph compounding both sensors? If we found ourselves in a **highly degraded environment** (such as a tunnel) and the LIO lost itself, what would happen to R3LIVE?

ziv-lin, 2022-02-07:
1. The LIO actually is affected by the VIO. After the update of VIO, the prior estimation of the LIO is propagated from the IMU propagation.
2. In a long tunnel-like environment, the overall system can still work normally by utilizing the measurement from the VIO subsystem (e.g., see Experiment-2 of our previous R2LIVE). 
[image]

Huguet57, 2022-02-07:
Thanks for the answer, now I see there's only one state (``g_lio_state``) being estimated both in the LIO subsystem and the VIO subsystem. That was what was confusing me.

Thanks for the great work

closed 2022-02-07 by Huguet57