hku-mars/r3live #51: [Issue] Steady inclination along SLAM

Camilochiang, 2022-02-10:
The data was captured 10 seconds after starting the algorithm, giving enough time for IMU calibration as previously recommended.
After post processing the 3D data and fitting a plane to the ground the following array result for the plane equation (ax + by +cz + d = 0):
`array([-0.05517862,  0.09160666,  0.99426533,  1.39617296])`

I do have the small impression that somewhere during the IMU initialization the value of Z get too high, this make all the 3D point cloud to have a slop of around 45° in the Z axis (simplified to 1 dimension).

ziv-lin, 2022-02-11:
What is your problem?

Camilochiang, 2022-02-11:
- The view angle is aligned to the Y axis
- The scanned place is known to be flat
- R3live estimate the full pointcloud tilded. When a plane is fitted the slope is almost 1, What i found weird no?

[image]

closed 2022-02-17 by ziv-lin

Camilochiang, 2022-02-21:
@ziv-lin , May I ask why this issue was closed?