hku-mars/r3live #52: lidar update

dongfangzhou1108, 2022-02-11:
[image]
大佬新年好！
论文里的公式是这个样子的，但是代码中
      solution = K * ( meas_vec - Hsub * vec.block< 6, 1 >( 0, 0 ) );
这里是不是把论文中的Jk近似为单位阵，但是论文中的I-KH，中的I好像代码中没对应上？
我是哪里理解错了呢？多谢！

dongfangzhou1108, 2022-02-11:
我将代码替换成了
`
Eigen::Matrix<double, Eigen::Dynamic, Eigen::Dynamic> Identity_DIM_OF_STATES;
Identity_DIM_OF_STATES = Eigen::MatrixXd::Identity(DIM_OF_STATES, DIM_OF_STATES);
Eigen::Matrix< double, DIM_OF_STATES, 1 > solution_;
solution_ = K * meas_vec + ( Identity_DIM_OF_STATES - K * Hsub ) * vec.block< 6, 1 >( 0, 0 );
`
使用您提供的hku_park_01.bag数据集，原始的和修正的都顺利回到了原点，没有明显的区别

ziv-lin, 2022-02-11:
Here we apply some equivalence for improving the runtime performance.

dongfangzhou1108, 2022-02-15:
您好，r3live/src/r3live_vio.cpp中，帧间pnp位姿估计部分，重投影误差雅各比部分，根据链导法则，将误差相对于位姿的雅各比，拆分为误差到三维点的雅各比和三维点相对于位姿的雅各比，您的误差到三维点雅各比的代码
mat_pre << fx / pt_3d_cam( 2 ), 0, -fx * pt_3d_cam( 0 ) / pt_3d_cam( 2 ), 0, fy / pt_3d_cam( 2 ), -fy * pt_3d_cam( 1 ) / pt_3d_cam( 2 );(703行)
是不是应该改为
mat_pre << fx / pt_3d_cam( 2 ), 0, -fx * pt_3d_cam( 0 ) / pow(pt_3d_cam( 2 ), 2), 0, fy / pt_3d_cam( 2 ), -fy * pt_3d_cam( 1 ) / pow(pt_3d_cam( 2 ), 2);

dongfangzhou1108, 2022-02-17:
大佬，R3LIVE::vio_photometric函数中，mat_photometric好像没有初始化赋值啊，这块应该是图像梯度把

dongfangzhou1108, 2022-02-17:
> 大佬，R3LIVE::vio_photometric函数中，mat_photometric好像没有初始化赋值啊，这块应该是图像梯度把

我已经debug过了，所有的solution的输出都是0

ziv-lin, 2022-02-17:
您好，感谢你的report，不排除是rlease代码的时候，从开发分支合并到这个分支出现的问题。我得找个时间确认下这个问题。
但香港这边疫情比较严重，实验室不开放了，我这阶段只能窝在家里改论文，等回到实验室我再检查下这个开源版本的这个代码问题。

dongfangzhou1108, 2022-02-17:
> 您好，感谢你的report，不排除是rlease代码的时候，从开发分支合并到这个分支出现的问题。我得找个时间确认下这个问题。 但香港这边疫情比较严重，实验室不开放了，我这阶段只能窝在家里改论文，等回到实验室我再检查下这个开源版本的这个代码问题。

多谢大佬，此外，针对白墙那个退化序列，我发现，注释map frame光度优化，仅仅保留光流pnp优化，是可以完美运行的，其他退化序列也是如此，然而，如果注释pnp优化，仅仅使用光度优化，那么退化序列将会严重漂移。
基于此，可以认为光流pnp对于退化场景的贡献更大，而仅仅使用光度误差却意义不大。是这样吗？

ziv-lin, 2022-02-17:
这点我得确认一下，不知道是不是有bug，我已经私信你1468012920@qq.com这个邮箱了，你能否看下？

ziv-lin, 2022-03-06:
@dongfangzhou1108 This bug have been fixed in the last commit f04db809922cceb646feb4837bdea0520b908a1f. 
See:
https://github.com/hku-mars/r3live/blob/f04db809922cceb646feb4837bdea0520b908a1f/r3live/src/r3live_vio.cpp#L885-L887
You can set DEBUG_PHOTOMETRIC = 1 in
https://github.com/hku-mars/r3live/blob/f04db809922cceb646feb4837bdea0520b908a1f/r3live/src/r3live_vio.cpp#L58
to check the result of r3live::vio_photometric
If you find any other problems, please report A.S.A.P

dongfangzhou1108, 2022-03-09:
> @dongfangzhou1108 This bug have been fixed in the last commit [f04db80](https://github.com/hku-mars/r3live/commit/f04db809922cceb646feb4837bdea0520b908a1f). See:
> 
> https://github.com/hku-mars/r3live/blob/f04db809922cceb646feb4837bdea0520b908a1f/r3live/src/r3live_vio.cpp#L885-L887
> 
> 
> You can set DEBUG_PHOTOMETRIC = 1 in
> https://github.com/hku-mars/r3live/blob/f04db809922cceb646feb4837bdea0520b908a1f/r3live/src/r3live_vio.cpp#L58
> 
> 
> to check the result of r3live::vio_photometric
> If you find any other problems, please report A.S.A.P

thanks ! all is right, best wishes

ziv-lin, 2022-03-11:
OK, glad to hear that this bug-fixing is confirmed.

closed 2022-03-11 by ziv-lin