hku-mars/r3live #58: 您好，我在运行自己的数据时无法显示图像，并且camera frame一直显示为2

ToutDonner, 2022-02-19:
[image]
[image]

ToutDonner, 2022-02-19:
[image]
这是我的标定结果

ToutDonner, 2022-02-19:
这是我标定的外参以及config中的配置
[image]

closed 2022-02-19 by ToutDonner

huangsiyuan0717, 2022-03-29:
@ToutDonner  你好，请问你是怎么解决的， 我也碰到了这个问题

Yangchunqiu, 2022-03-31:
这个应该就是camera的数据没接入进去，尝试着在launch file里面把相机的topic改成自己的

ToutDonner, 2022-03-31:
> 有可能是相机和雷达时间不同步，使用他们修改过后的livox driver 应该是以系统时间为标准的
> 
> > 这个应该就是camera的数据没接入进去，尝试着在launch file里面把相机的topic改成自己的
> 
> 
> 
> 我也出现这个情况 改了自己的话题还是只有两帧(相机用的realsense d455)
> 
>

duckbubi29, 2022-04-27:
> > 有可能是相机和雷达时间不同步，使用他们修改过后的livox driver 应该是以系统时间为标准的
> > > 这个应该就是camera的数据没接入进去，尝试着在launch file里面把相机的topic改成自己的
> > 
> > 
> > 我也出现这个情况 改了自己的话题还是只有两帧(相机用的realsense d455)

你好，怎么解决时间不同步问题呢

FengYuQ, 2022-06-02:
请问这个问题解决了嘛

ToutDonner, 2022-06-04:
解决了，我这里是因为雷达和相机时间没有同步，要使用修改过后的livox driver

wanghui-min, 2022-07-27:
> 解决了，我这里是因为雷达和相机时间没有同步，要使用修改过后的livox driver

能告知怎么进行同步吗？或者livoxdriver哪里修改？

ToutDonner, 2022-07-27:
> > 解决了，我这里是因为雷达和相机时间没有同步，要使用修改过后的livox driver
> 
> 
> 
> 能告知怎么进行同步吗？或者livoxdriver哪里修改？

直接使用mars他们修改的那个就可以，在他们github里有

Russell-fan, 2022-09-06:
请问一下，是需要使用作者修改的那个[livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE)录制包吗？我用的是这个修改过的驱动录制的包，可还是有这个错误。

ToutDonner, 2022-09-06:
我是这么做的，你可能需要调整一下相机？> 请问一下，是需要使用作者修改的那个[livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE)录制包吗？我用的是这个修改过的驱动录制的包，可还是有这个错误。

ToutDonner, 2022-09-06:
> 请问一下，是需要使用作者修改的那个[livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE)录制包吗？我用的是这个修改过的驱动录制的包，可还是有这个错误。

我是这么做的，你可能需要检查一下相机？

farhad-dalirani, 2023-02-04:
Hello,

I use Realsense, and I have a similar problem. Can you explain how you solved it?

https://github.com/hku-mars/r3live/issues/156#issue-1570393194

farhad-dalirani, 2023-02-04:
> > Excuse me, do I need to use the [livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE) recording package modified by the author? I use this modified driver recording package, but I still have this error.
> 
> That's what I did, you might want to check the camera?

Hello @ToutDonner ,

I use Realsense, and I have a similar problem. Can you explain how you solved it?

https://github.com/hku-mars/r3live/issues/156#issue-1570393194

FengYuQ, 2023-02-04:
What's your OpenCV version, I remember when I change the opencv version to 3.14  the program was worked.

冯宇祺
***@***.***

   

  
  
 
 
#156 (comment)

ToutDonner, 2023-02-04:
> > > Excuse me, do I need to use the [livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE) recording package modified by the author? I use this modified driver recording package, but I still have this error.
> 
> > 
> 
> > That's what I did, you might want to check the camera?
> 
> 
> 
> 
> 
> Hello @ToutDonner ,
> 
> 
> 
> I use Realsense, and I have a similar problem. Can you explain how you solved it? Also, I used Velodyne LiDAR. You can see my problem here:
> 
> 
> 
> https://github.com/hku-mars/r3live/issues/156#issue-1570393194

@farhad-dalirani you can record a piece of data and check the time reference of the camera and the lidar. r3live does not need hard synchronization, but the two sensors need to use the same time reference

farhad-dalirani, 2023-02-04:
> > > > Excuse me, do I need to use the [livox_ros_driver_for_R2LIVE](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE) recording package modified by the author? I use this modified driver recording package, but I still have this error.
> > 
> > 
> > > 
> > 
> > 
> > > That's what I did, you might want to check the camera?
> > 
> > 
> > Hello @ToutDonner ,
> > I use Realsense, and I have a similar problem. Can you explain how you solved it? Also, I used Velodyne LiDAR. You can see my problem here:
> > [#156 (comment)](https://github.com/hku-mars/r3live/issues/156#issue-1570393194)
> 
> @farhad-dalirani you can record a piece of data and check the time reference of the camera and the lidar. r3live does not need hard synchronization, but the two sensors need to use the same time reference

@ToutDonner Thank you for the quick reply, I really appreciate it.

What do you mean by the same-time reference? For example, if both camera and Lidar messages have ROS time in their header, does it satisfy your comment?

ToutDonner, 2023-02-04:
> @ToutDonner Thank you for the quick reply, I really appreciate it.
>
>
>
> What do you mean by the same-time reference? For example, if both camera and Lidar messages have ROS time in their header, does [...]

You can look this pic[image]

farhad-dalirani, 2023-02-09:
Hi, @ToutDonner and @FengYuQ 

Hi, since I use Velodyne and a camera and mark message with ROS, it seems there is no synchronization problem. However, I experience a significant drift. I explained my problem in this issue:

https://github.com/hku-mars/r3live/issues/157

If you look at it, it would be great.