hku-mars/r3live #61: std::bad_alloc

RobotGithubLee, 2022-02-21:
when run "roslaunch r3live r3live_bag_ouster.launch"

=============================================================
App name   : R3LIVE: A Robust, Real-time, RGB-colored, LiDAR-Inertial-Visual tightly-coupled state Estimation and mapping package
Build date : Feb 21 2022  17:49:41
CPU infos  : Intel(R) Core(TM) i7-10700K CPU @ 3.80GHz
RAM infos  : 15.49GB Physical Memory 977.00MB Virtual Memory
OS  infos  : Linux 4.15.0-142-generic (x86_64)
Home dir   : /home/leo/
Current dir: /home/leo/.ros/
Date mow   : 2022/02/21 17:58:53
=============================================================
+++++++++++++++++++++++++++++++++++++++++++++++
Here is the your software environments: 
GCC version          : 5.4.0
Boost version        : 1.58.0
Eigen version        : 3.3.9
OpenCV version       : 3.3.1
+++++++++++++++++++++++++++++++++++++++++++++++
Multi thread started 
terminate called after throwing an instance of 'std::bad_alloc'
  what():  std::bad_alloc
================================================================================REQUIRED process [r3live_mapping-2] has died!

RobotGithubLee, 2022-02-22:
I set "m_rgb_pts_vec.reserve( 1e7 )"

use bag "degenerate_seq_00.bag"
 
when Bag Time: 1629515038.933057   Duration: 40.724580 / 101.880594

| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   16:19:49  |    60       |    195       | 182021       |    540          /home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
/home/leo/r3live_ws/src/r3live/r3live/src/loam/IMU_Processing.cpp, 132, check_state fail !!!! -7.18497  1.75922  12.9929
state_inout |[1629515004.16395] | (21.409, -32.488, -6.057) | (-0.082, -0.570, 1.905) | (-7.185, 1.759, 0.000) | (0.164, 5.667, -0.579) | (2.702, -1.475, -2.179) 
state_in |[1629515004.16395] | (21.409, -32.488, -6.057) | (-0.082, -0.570, 1.905) | (-7.185, 1.759, 12.993) | (0.164, 5.667, -0.579) | (2.702, -1.475, -2.179) 
| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |

closed 2022-03-09 by stale[bot]

hr2894235132, 2022-10-27:
> I set "m_rgb_pts_vec.reserve( 1e7 )"
>
> use bag "degenerate_seq_00.bag"
>
> when Bag Time: 1629515038.933057 Duration: 40.724580 / 101.880594
>
> | System-time | LiDAR-frame | Camera-frame | Pts in maps | [...]

Hello, did you solve the problem?I have the same problem