hku-mars/r3live #62: [r3live_LiDAR_front_end-2]

MOSEAA, 2022-02-25:
你好，我遇到了一个很奇怪的问题，我重新装了多遍opencv，依然有这样的问题，请问您可以给我一些建议吗？真的好难搞哟
对了，我的rqt_image_view也可以正常使用。

[image]

MOSEAA, 2022-02-25:
[image]

MOSEAA, 2022-02-25:
[image]

zengkailiang, 2022-03-07:
> 我找到问题所在了，尝试多次，与环境配置有关，我的环境配置是ubuntu18.04 + gcc7.5 + boost 1.65.1 + eigen 3.3.4 + opencv 3.4.16

我也遇到类似的问题，我的环境是ubuntu18.04 + gcc7.5 + boost 1.65.1 + eigen 3.3.4 + opencv 4.4.0 
================================================================================REQUIRED process [r3live_mapping-3] has died!
process has died [pid 17396, exit code -11, cmd /home/lzk/r3live/devel/lib/r3live/r3live_mapping __name:=r3live_mapping __log:=/home/lzk/.ros/log/b122e9a4-9de4-11ec-9fe8-b42e99818be9/r3live_mapping-3.log].
log file: /home/lzk/.ros/log/b122e9a4-9de4-11ec-9fe8-b42e99818be9/r3live_mapping-3*.log
Initiating shutdown!
================================================================================

你是如何解决的呢

zengkailiang, 2022-03-07:
> 我找到问题所在了，尝试多次，与环境配置有关，我的环境配置是ubuntu18.04 + gcc7.5 + boost 1.65.1 + eigen 3.3.4 + opencv 3.4.16

我也遇到类似的问题, 我的环境是 ubuntu18.04 + gcc7.5 + boost 1.65.1 + eigen 3.3.4 + opencv
================================================================================REQUIRED process [r3live_mapping-3] has died!
process has died [pid 17396, exit code -11, cmd /home/lzk/r3live/devel/lib/r3live/r3live_mapping __name:=r3live_mapping __log:=/home/lzk/.ros/log/b122e9a4-9de4-11ec-9fe8-b42e99818be9/r3live_mapping-3.log].
log file: /home/lzk/.ros/log/b122e9a4-9de4-11ec-9fe8-b42e99818be9/r3live_mapping-3*.log
Initiating shutdown!
================================================================================
你是怎么解决的呢

closed 2022-03-07 by MOSEAA