hku-mars/r3live #63: How did you record your .bag file?

BellosC, 2022-03-02:
Hello, could you **share the .launch file** that you used in order to get the following .bag file *(hku_campus_seq_00.bag)*?
[image]
*(I am trying to collect my own data but for the moment i am not sure how to do that)*
Thank you in advance.

ziv-lin, 2022-03-11:
I record the rosbag file with the following command:
```
rosbag record /camera/image_color/compressed /camera/image_color/compressed/parameter_descriptions /camera/image_color/compressed/parameter_updates /livox/imu  /livox/lidar
```
Of course, you can also record all ros messages with:
```
rosbag record -a
```

BellosC, 2022-03-11:
Hi, 

Thanks for your feedback.

Although my question was not how you record the bag file.

My question was how you triggered the livox and the camera to start recording data simultaneously.

If you could share the launch file for that, it would be extremely helpful.

Thanks again!

closed 2022-03-26 by stale[bot]