hku-mars/r3live #66: Issue with camera calibration parameters

aditdoshi333, 2022-03-08:
Hello @ziv-lin,

I am trying r3live with a new Basler camera but facing some weird issues with RGB calibration. I calibrated RGB and livox using (https://github.com/hku-mars/livox_camera_calib) and the color point cloud came very well-calibrated. But when I am using the same matrix in the r3live config I am getting some static offset. Ideally, what should be image input? Rectified or distorted?

Example:
Output after calibration
[image]

Output of r3live

[image]

aditdoshi333, 2022-03-09:
@ziv-lin is it a problem for r3live if the fov of the camera is larger than fov of lidar?

zaalsabb, 2022-03-09:
For me I had to invert the extrinsics matrix I got from [livox_camera_calib](https://github.com/hku-mars/livox_camera_calib) before I put it in r3live config.

BellosC, 2022-03-17:
@aditdoshi333 could you please provide a link of **the Basler camera that you chose**? 

I am thinking of buying a model that supports GigE Vision 2.0 but i am not entirely sure if it is the right think to do.
My goal is to achieve the easiest possible timesync between camera and livox.

Thank you in advance for your time, and sorry if my comment is considered off-topic.

tdcsu, 2024-05-09:
> For me I had to invert the extrinsics matrix I got from [livox_camera_calib](https://github.com/hku-mars/livox_camera_calib) before I put it in r3live config.

for me, the martrix was invterted but there exist some static offset