hku-mars/r3live #67: show tracked points in image

zaalsabb, 2022-03-09:
Hi, how can we visualize the tracked points on the image similar to this example?
https://github.com/ziv-lin/r3live_dataset/blob/main/gifs/degenerate_00.gif

When I run the example I don't see the tracked points in rviz in the /track_img topic. How do I visualize the tracked points?

Thanks!

ziv-lin, 2022-03-11:
This part of the debugging code has been removed in this release version. You can visualize the tracked points by plotting the lines among tracked points in the current-frame and last-frame.

closed 2022-03-26 by stale[bot]

ly-uuu, 2022-11-28:
> Hi, how can we visualize the tracked points on the image similar to this example? https://github.com/ziv-lin/r3live_dataset/blob/main/gifs/degenerate_00.gif
> 
> When I run the example I don't see the tracked points in rviz in the /track_img topic. How do I visualize the tracked points?
> 
> Thanks!
Did you solve it?