hku-mars/r3live #68: livox lidar data type error

gongyue666, 2022-03-11:
Hi, I use my livox-mid70 and realsense d435 to run r3live, the lidar data type is always. when I use sensor_msgs/PointCloud2, it told me should use livox/cutom, and  when I change the type, it told me need sensor_msgs ....error logs is like this:
[ERROR] [1646964020.047520342]: Client [/r3live_LiDAR_front_end] wants topic /livox/lidar to have datatype/md5sum [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6], but our version has [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181]. Dropping connection.

[ERROR] [1646963951.610199012]: Client [/r3live_mapping] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.
---

My system is ubuntu 20.04, livox_driver is from r2live version.
here is config file:
Lidar_front_end:
   lidar_type: 1   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 6
   using_raw_point: 1
   point_step: 1
   
r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 4  # The point step of append point to global map. [default = 4]

r3live_vio:
   image_width: 1280
   image_height: 720
   camera_intrinsic:
      [916.984038571582, 0, 641.0356103230023,
0, 920.3167095605404, 359.3140181951169,
0, 0, 1 ] 
   camera_dist_coeffs: [0,0,0,0,0]  #k1, k2, p1, p2, k3
   # Fine extrinsic value. form camera-LiDAR calibration.
   camera_ext_R:
         [0.0110199,  -0.999796,  0.016932 ,
          -0.24723,  -0.0191315,  -0.968768,
          0.968894,  0.00648958,  -0.247391]
   # camera_ext_t: [0.050166, 0.0474116, -0.0312415] 
   camera_ext_t: [ 0.0818775,0.0398223,-0.0745301] 
   # Rough extrinsic value, form CAD model, is not correct enough, but can be online calibrated in our datasets.
   # camera_ext_R:
   #    [0, 0, 1,
   #     -1, 0, 0,
   #     0, -1, 0]
   # camera_ext_t: [0,0,0] 
   
r3live_lio:        
   lio_update_point_step: 4   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: 0        # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30

and here is launch file:
<launch>
    
    <param name="LiDAR_pointcloud_topic" type="string" value= "/livox/lidar" />
    <param name="IMU_topic" type="string" value= "/IMU_data" />
    <param name="Image_topic" type="string" value= "/camera/color/image_raw" />
    <param name="r3live_common/map_output_dir" type="string" value="$(env HOME)/r3live_output" />
    <rosparam command="load" file="$(find r3live)/../config/giraffe_config.yaml" />

    <node pkg="r3live" type="r3live_LiDAR_front_end" name="r3live_LiDAR_front_end"  output="screen" required="true"/>
    <node pkg="r3live" type="r3live_mapping" name="r3live_mapping" output="screen" required="true" />
    
    <arg name="rviz" default="1" />
    <group if="$(arg rviz)">
        <node name="rvizvisualisation" pkg="rviz" type="rviz" output="log" args="-d $(find r3live)/../config/rviz/r3live_rviz_config.rviz" />
    </group>
</launch>

Camilochiang, 2022-03-11:
Well, basically is what is saying. You are using the wrong LiDAR message type. Did you pre recorded the rosbag?

also, if you are using mid70 you can change the parameter lidar_front_end / scans from 6 to 1 has mid70 have only one line at diferente of avia

ziv-lin, 2022-03-11:
Yes, I agree with @Camilochiang, I guess you are using the wrong LiDAR message type. Are your using the PCL_pointcloud2? the right one is livox_custom_message (see our example).

gongyue666, 2022-03-14:
[image]
acturelly I use the message type as livox_ros_driver/CustomMsg, but it also get this error. No matter I use "livox_ros_driver/CustomMsg" or "sensor_msgs/PointCloud2" I will get the error that need the other message the type. Another error  log:
[image]

gongyue666, 2022-03-14:
Do the node  "r3live_mapping" and node "r3live_Lidar_front_end" need different data type?

closed 2022-03-15 by gongyue666

ruiche01, 2022-04-11:
@gongyue666 Do you solve this problem? I also meet this problem

gongyue666, 2022-04-13:
check the launch file, the lidar data type is different in the two nodes

linzs-online, 2022-11-25:
I've met the same problem , couold you tell me how you fixed this problem? @gongyue666 specifically where you changed the param .thanks.

jiatianluo, 2024-04-20:
请问您解决这个问题了吗，我也遇到了同样问题

peter-ivarsen, 2024-06-26:
If you have a bag with lidar messages on the sensor_msgs/PointCloud2 format, you could set lidar_type: 0 in r3live_config.yaml.
This will activate the MID40 lidar type in LiDAR_front_end.cpp wich uses ha handler that handles sensor_msgs/PointCloud2