hku-mars/r3live #71: r3live runs on arm architecture

tgj891, 2022-03-16:
hi,
   I want to run r3live on jetson agx Xavier, but I get an error: cpuid.h not found, and 'c++ unrecognized command line option -msse2 -msse3', how to solve this? Still it doesn't currently support running on arm.

Camilochiang, 2022-03-31:
Hei @ziv-lin any news about this? was also mentioned in #32, and it will be nice to know about the road plan of your lab!
Thanks!

LangDaoAI, 2022-06-23:
@ziv-lin 林博， 这个有任何更新或者如何改造的建议吗？

rlabs-oss, 2022-07-07:
doesnt seem to be the case, it uses cpuid, and e.g src/r3live/r3live/src/optical_flow/lkpyramid.cpp:377:13: error: ‘qw1’ was not declared in this scope
  377 |             qw1 = _mm_set1_epi32(iw10 + (iw11 << 16));

which doesnt build on aarch64.

JasonWang9805, 2022-09-02:
现在解决了吗？我看到文件中对sse相关代码都有提示

closed 2022-09-17 by stale[bot]