hku-mars/r3live #75: image topic error

huangsiyuan0717, 2022-03-23:
I follow readme.md to run my data. Meeting the follow err.(camera: realsense d455)
[image]

zengkailiang, 2022-04-02:
image和lidar时间戳不同步导致的吧

closed 2022-04-08 by huangsiyuan0717

duckbubi29, 2022-04-27:
> Thanks for your great job! I try to run my onw data, but just can capture 2 camera frame, what can I do? My environment as the following:(hardware: livox-horizon, realsense D455 [image] [image] [image] And I changed the configure `<param name="Image_topic" type="string" value= "/camera/color/image_raw" />` Also there are other param changed in r3live.yaml. Maybe the same as [#58 (comment)](https://github.com/hku-mars/r3live/issues/58#issue-1144605301)

你好，我也遇到了相同的问题，请问你解决了吗

farhad-dalirani, 2023-02-04:
@huangsiyuan0717 @zengkailiang Did you solve the problem? I use velodyne-16 and a realsense.