hku-mars/r3live #78: imu出现路径漂移问题

fl840249479, 2022-04-07:
您好！我的静止不动运行r3live时能顺利生成正确点云，但对设备进行微小位移时就会出现大幅度路径漂移问题。如图，仅仅移动了一点点r3live就显示路径就开始乱窜，请问怎么解决呢。雷达使用的是livox-mid40，没有自带imu，相机使用的是azura kinect dk，imu使用的是kinect的imu
[image]

fl840249479, 2022-04-07:
> I had that issue many times. The problem is that r3live is not writen for external IMU (as your case). You have to modify the code (including Livox ros driver)

十分感谢您的回复。您是否可以告诉我们应该修改哪些代码的哪些部分呢？以及您那边是否有修改过的代码可供参考呢？
Thanks for your reply. Could you please tell us in detail which part of the code should be modified? And would you like to share your modified code if you have?

Camilochiang, 2022-04-07:
Sorry. You actually have to only modify livox_ros_driver. You can find more info [here](https://github.com/hku-mars/r2live/issues/56) or download my [repository](https://github.com/Camilochiang/livox_ros_driver_external_IMU)

ly-cao, 2022-04-07:
> 
我有一个疑问，就是lio_config.yaml 中找不到像“extrinsic_T”或“extrinsic_R”这样的参数，那么如何在 LiDAR 和 IMU 之间设置外部参数呢？

vipelz, 2022-04-07:
I also have a mid40. Could you please share wich IMU model do you have and how to connect? 
Thank you a lot

ly-cao, 2022-04-08:
> 

realsense d435i （camera with imu）

Camilochiang, 2022-04-09:
@fl840249479  Please close the topic if your question has been answer ;)

fl840249479, 2022-04-10:
> @fl840249479 Please close the topic if your question has been answer ;)

非常抱歉这么晚回复您。我们按照您提供的代码修改了livox驱动后有改善情况但并没有完全解决路径飘移问题。当我们手持设备仅仅进行一小段移动时，路径在r3live中会出现大幅度的错误的飘移（像我问题里发出的图），导致map结果不准确。
I'm very sorry to reply you so late. We modified the livox_ros_driver according to the code you provided, which improved the situation, but did not completely solve the problem of path drift. When our  device only moves for a short period, the path will drift significantly wrong in r3live (like the figure sent out in my question), resulting in inaccurate map results.

closed 2022-04-25 by stale[bot]

Tomato1107, 2022-10-04:
> 非常抱歉这么晚回复您。我们按照您提供的代码修改了livox驱动后有改善情况但并没有完全解决路径飘移问题。当我们手持设备仅仅进行一小段移动时，路径在r3live中会出现大幅度的错误的飘移（像我问题里发出的图），导致map结果不准确。 I'm very sorry to reply you so late. We modified the livox_ros_driver according to [...]

因为直接用d435i的imu数据的frame_id是camera_imu_optical_frame，朝向与livox的lidar是不一致的。可以重新写个imu的订阅预发布的程序使得imu输出数据的朝向与lidar一致（或者使用tf转换发布imu在camera_link的frame_id下的imu数据），然后就能顺利跑r3live了

farhad-dalirani, 2023-02-09:
@fl840249479 @Camilochiang @Tomato1107 

How did you solve the drifting problem? I have a drifting problem with Velodyne Lidar+IMU+RealSense camera.
I explained it in detail. It would be great if look at it:

https://github.com/hku-mars/r3live/issues/157

fanshixiong, 2023-03-04:
你好，非常感谢你们的工作，我在使用r3live时出现了很大的漂移，尤其是转弯处，直线处效果还行，但是同样的参数配置在fast-lio2里却取得了很好的效果。
我的r3live版本是添加了外置imu旋转的版本，仓库在这里：@tiny442 https://github.com/tlin442/r3live
我使用的激光雷达是livox mid-70，外置imu，使用的是mynt-D1000相机和imu，同时使用了修改过后的livox driver。
我的电脑是ubuntu20.04。
这是我的配置文件：
```
Lidar_front_end:
   lidar_type: 1   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 4
   using_raw_point: 1
   point_step: 1
   min_intensity: 20.0
   min_dist: 0.1
   lidar_imu_rotm:
      # LiDAR is mounted rotated by 90 deg
      #[1, 0, 0,
      # 0, 0, 1,
      # 0, -1, 0]
      [ 0.020712, -0.999783,  0.002253,
   0.073848,  0.003778,  0.997262,
  -0.997054, -0.020489,  0.073910]
   lidar_imu_tranm: 
      [0.037628, 0.067744, 0.098026]

r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 20  # The point step of append point to global map. [default = 4]

r3live_vio:
   
   # This is the real camera intrinsics used for computing the undistort map for fisheye
   camera_intrinsic_real:
      [655.005, 0, 679.029, 
      0, 656.097, 358.596, 
      0, 0, 1 ] 
   # This is the projection intrinsics used for the rest of the calculations
   camera_intrinsic:
      [655.005, 0, 679.029, 
      0, 656.097, 358.596, 
      0, 0, 1 ] 
      # [698.4 , 0, 650.593, 
      # 0, 698.4, 359.906, 
      # 0, 0, 1 ] 
   # Hacked in r3live_vio.cpp to be fisheye instead of normal pinhole
   camera_dist_coeffs: [-0.238605, 0.0435143, 0.000366211, -0.00272751, 0]  #k1, k2, p1, p2, k3
  # Camera->IMU value derived from T265 factory calibration.
   camera_ext_R:
         [0.886267, -0.0596407,  -0.459319,
         0.0143203, -0.987673,   0.155877,
         -0.462953, -0.144726,  -0.874488]
         #[0, 0, 1,
         # -1, 0, 0,
         # 0, -1, 0]
   #camera_ext_t: [-0.013776, 0.0347174, -0.0972432] 
   camera_ext_t: [ 0, 0, 0 ]
   
r3live_lio:        
   lio_update_point_step: 6   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: -0.092132        # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30
   publish_feature_map: 1
```
这是launch文件：
```
<launch>
    <!-- Subscribed topics -->
    <param name="LiDAR_pointcloud_topic" type="string" value="laser_cloud_flat" />
    <param name="IMU_topic" type="string" value= "/mynteye/imu/data_raw" />
    <param name="Image_topic" type="string" value= "/mynteye/right/image_color" />
    <param name="r3live_common/map_output_dir" type="string" value="$(env HOME)/r3live_output" />
    <rosparam command="load" file="$(find r3live)/../config/r3live_mid70_t265.yaml" />

    <node pkg="r3live" type="r3live_LiDAR_front_end" name="r3live_LiDAR_front_end"  output="screen" required="true"/>
    <node pkg="r3live" type="r3live_mapping" name="r3live_mapping" output="screen" required="true" />
    
    <arg name="rviz" default="1" />
    <group if="$(arg rviz)">
        <node name="rvizvisualisation" pkg="rviz" type="rviz" output="log" args="-d $(find r3live)/../config/rviz/r3live_rviz_config.rviz" />
    </group>
</launch>
```
请问有哪些地方有问题呢，谢谢回复，

farhad-dalirani, 2023-03-06:
@fanshixiong Hi, Can you explain your setup, Camera type, lidar, positioning of IMU, etc.

fanshixiong, 2023-03-30:
@farhad-dalirani sorry for seeing it so late
My r3live version is the version with external imu rotation added, the warehouse is here: @tiny442 https://github.com/tlin442/r3live
The lidar I use is livox mid-70 with external imu, mynt-D1000 camera and imu, and the modified livox driver.
My computer is ubuntu20.04.
Here is my config file:
```
Lidar_front_end:
   lidar_type: 1   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 6
   using_raw_point: 1
   point_step: 1
   lidar_imu_rotm:
      # LiDAR is mounted rotated by 90 deg
      #[1, 0, 0,
      # 0, 0, 1,
      # 0, -1, 0]
      [ 0.016511, -0.999700,  0.018083,
       0.057071,  0.018999,  0.998189,
       -0.998234, -0.015449,  0.057368]
   lidar_imu_tranm: 
      [0.039342, 0.077608, 0.037443]

r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 4  # The point step of append point to global map. [default = 4]

   res_path: "/home/frans/code/r3live_proj/catkin_ws_r3live/src/r3live_/res"

r3live_vio:
   image_width: 1280
   image_height: 720
   camera_intrinsic:
       [655.005, 0, 679.029,
       0, 656.097, 358.596,
      0, 0, 1]
   camera_dist_coeffs: [-0.238605, 0.0435143, 0.000366211, -0.00272751, 0]  #k1, k2, p1, p2, k3
   
   # Fine extrinsic value. form imu2camera calibration.
   camera_ext_R:
         [0.999998,  0.00183758, 0.000849753,
         0.00184018,   -0.999994, -0.00307635,
         0.000844095,  0.00307791,   -0.999995]
   camera_ext_t: [0.0993128, 0.0117891, -0.176605] 

   
r3live_lio:        
   lio_update_point_step: 4   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: -0.092132       # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30

```

The imu_tools used for the internal reference calibration of the imu, radar and imu use is your job:[hku-mars/LiDAR_IMU_Init](https://github.com/hku-mars/LiDAR_IMU_Init), the camera and the imu use kalibr calibration, and the reprojection error is about 1.5 pixels.
[image]

The test results on fast-lio2 are better, and the test results on vins are average, but there are obvious drifts in r3live.
Is there any good solution to solve the drift? Thanks for the reply.
[image]
[image]
There is a large drift when facing the wall, but there is no drift in fast-lio2.
Thanks.