hku-mars/r3live #90: 您好，我这边测试r3live的r3live_reconstruct_mesh.launch时程序会挂掉，不知道您有没有遇到过这种情况？

dandelion-chu, 2022-05-13:
我的环境如下图所示，opencv使用版本是4.5.3，ubuntu18.04 melodic，运行roslaunch r3live r3live_bag.launch正常，并且使用hkust_campus_seq_00.bag成功跑出来了建图且保存为pcd文件了；但是测试roslaunch r3live r3live_reconstruct_mesh.launch时，出现问题，程序加载完test.r3live之后就挂掉了，不知道是怎么回事，谢谢哈~~

[image]
[image]

nicetryzz, 2022-05-15:
memory太小了

dandelion-chu, 2022-05-16:
> 

好的，我目前是配置了8G内存和8G SWAP，后面升级下硬件再试试，谢谢哈~~~

dandelion-chu, 2022-05-16:
> memory太小了

好的，我目前的配置是8G RAM+8G SWAP，后面升级下硬件再重新试一下

dandelion-chu, 2022-05-31:
> memory太小了

您好，我现在升级设备重新测试，mesh程序还是会挂掉，请问能帮忙排查下是什么问题嘛？我的内存是16G，SWAP是8GB，谢谢哈~~
[image]
[image]
[image]

closed 2022-06-15 by stale[bot]

abnerlam, 2022-09-28:
@dandelion-chu 请问您解决了吗？