hku-mars/r3live #94: velodyne lidar

qqstar-423, 2022-05-21:
请问大家有使用旋转雷达运行过吗，我使用的是速腾的32线雷达，模仿fastlio在r3live中写了一个velodyne的回调函数，但是运行起来效果不是很好，很容易产生退化。
[image]
这是我的运行界面，里程计已经发生退化

WillBao33, 2022-06-05:
> 请问大家有使用旋转雷达运行过吗，我使用的是速腾的32线雷达，模仿fastlio在r3live中写了一个velodyne的回调函数，但是运行起来效果不是很好，很容易产生退化。 [image] 这是我的运行界面，里程计已经发生退化

你好，请问能分享一下您是如何用velodyne雷达运行的吗？ 谢谢

Fellowflyingsky, 2022-12-18:
> > [image]问有过吗?吗?吗?吗?吗的的的的的的的的的的的的的的的的的的的的的的的的的的的的的吗？
> 
> 你好，请问能分享一下您是如何使用velodyne雷达运行的吗？谢谢

你好，请问你解决机械雷达和相机的这个问题了么

farhad-dalirani, 2023-01-20:
https://github.com/hku-mars/r3live/issues/94#issue-1243960702

Would you please share your code for adopting Velodyne?

farhad-dalirani, 2023-02-04:
I have the same problem, could you solve it?

farhad-dalirani, 2023-02-09:
Hi @qqstar-423 and @Fellowflyingsky ,

Could you solve the problem? I have similar problem and I explained it in detail in the following issue:

https://github.com/hku-mars/r3live/issues/157

If you look at it, it would be great.

Thanks.

jingyilon, 2024-03-13:
> 请问大家有使用旋转雷达运行过吗,我使用的是速腾的32线雷达,模仿法斯利奥在r3live中写了一个威力登的回调函数,但是运行起来效果不是很好,很容易产生退化。 [image]这是我的运行界面,里程计已经发生退化

您好，我刚了解这些，能开源一下您的算法么，非常感谢2427375498@qq.com

maffan-96, 2024-05-13:
did you get 

> > Have you ever used a rotating radar to run it? I used Sagitar's 32-line radar. I imitated Faslio and wrote a Velodyne callback function in r3live, but the running effect is not very good and it is easy to degrade. [image]This is my running interface, the odometer has degraded
> 
> Hello, I just learned about this. Can you open source your algorithm? Thank you very much [2427375498@qq.com](mailto:2427375498@qq.com)

Hey, would you be kind enough to share the Velodyne handler code if you received it?