hku-mars/r3live #106: Global optimization

antun008, 2022-06-28:
Can you put in algorithm global optimization (least sqaure adjustment) I think with this ee can get even better result.

Camilochiang, 2022-06-28:
> Can you put in algorithm global optimization (least sqaure adjustment) I think with this ee can get even better result.

Can you add a reference please?

antun008, 2022-06-28:
something like bundle adjustment what using sfm algorithm: it uses equalization of all images at once and of course it's all done in post processing.  I know the purpose of SLAM algorithms is to be real-time, but when viewed from a r3live perspective, it's a mapping package and then the same thing could be done in it, using all the frames covered by lidar and the camera.  Do not look at all to reduce the required computational resources, because post-processing can be done on a strong computer.  Optimization could be done based on identical features, with added loop closure (bag of words and scan context).  Of course, everything would work even better if a gps sensor were added.  When I look at it completely, r3live is a really powerful algorithm, which could become even better :)

Camilochiang, 2022-06-29:
Well. What you are asking for is more like a "patch" for R3live as could work in post processing.  
A loop detection implementation should be relatively easy to implement using [loam_livox](https://github.com/hku-mars/loam_livox) as a base.  Another options will be to have a look to [FAST_LIO_SLAM](https://github.com/gisbi-kim/FAST_LIO_SLAM), [FAST_LIO_LC](https://github.com/yanliang-wang/FAST_LIO_LC) who also implement loop closure

antun008, 2022-06-29:
> Well. What you are asking for is more like a "patch" for R3live as could work in post processing.
> A loop detection implementation should be relatively easy to implement using [...]

I agree with you, but really don't know how to implement this in r3live.

antun008, 2022-06-29:
I would like to show the results I got with R3Live. The delay is happening because there is no additional loop closure. Maybe I’m looking at it in a layman’s way because I don’t know how to implement it, I just want to give you maybe some idea.
Here is one example 1920.pcd, both walls should be at the same height, exterior and interior:
https://geofhr-my.sharepoint.com/:u:/g/personal/ajakopec_geof_hr/ER7Yi3uEk-9Di2k6BCXQIBcB-lsbUvtVgY3dh_oZyVOSAQ?e=ExrAWu

[image]
[image]
[image]

Same thing with hnk.pcd, when I went back to the beginning on my own, I got double walls and when the wall was taken a second time, a height difference of 3 meters was obtained:
https://geofhr-my.sharepoint.com/:u:/g/personal/ajakopec_geof_hr/ERA7iir-mf5Oh8q9H2-3UwwBYkb1yxeVbTgj42NyqklbJA?e=Fh6Glw

[image]

antun008, 2022-06-30:
Maybe reconfigure BALM to get bundle adjustment for r3live

zaalsabb, 2022-06-30:
Agreed. This would be a really useful feature!

antun008, 2022-07-04:
Here is one more example, where I get double walls:
[image]
[image]
https://geofhr-my.sharepoint.com/:u:/g/personal/ajakopec_geof_hr/EeyZcIa2zw9LloIOCG9yj30B2hgK0Whdj0I_s2AtnYQ5wg?e=GOJElX

closed 2022-07-20 by stale[bot]

antun008, 2022-09-27:
> something like bundle adjustment what using sfm algorithm: it uses equalization of all images at once and of course it's all done in post processing.  I know the purpose of SLAM algorithms is to be real-time, but when viewed from a r3live perspective, it's a mapping package and then the same thing could be done in it, using all the frames covered by lidar and the camera.  Do not look at all to reduce the required computational resources, because post-processing can be done on a strong computer.  Optimization could be done based on identical features and using gtsam optimizer or similirar, with added loop closure and detection (bag of words - DBoW2 and scan context).  Of course, everything would work even better if a gps sensor were added.  When I look at it completely, r3live is a really powerful algorithm, which could become even better :)
> Like google cartographer, it has pose adjustment, or some similar think. 
> [image]
> I think we should look at the way that the algorithm does not "forget" the previous positions of the sensors, but rather takes everything into account (all poses) in order to get connection between all 3d points.
>  relatively better.
> Here is reference to graph based optimization: https://www.google.com/url?sa=t&source=web&rct=j&url=http://www2.informatik.uni-freiburg.de/~stachnis/pdf/grisetti10titsmag.pdf&ved=2ahUKEwjt9f6D19D4AhUKIMUKHRUOBIsQFnoECAUQAQ&usg=AOvVaw3_l9o1dOk82MwBGeOO-eE4

Camilochiang, 2022-09-27:
Have a look to [BALM 2.](https://github.com/hku-mars/BALM)