hku-mars/r3live #107: Large point cloud map twisting issue with vio enabled

WillBao33, 2022-06-28:
Hi,

Thank you for your work, it's very impressive. 
I tried to run the program with Velodyne 16 lidar, using the velo16_handler from R2LIVE. When I enable the vio, the point cloud map looks twisted and the RGB image is also not properly showing on the point cloud map, as you can see in the picture below. I also got the "Failed to find match for field 'normal_x,y,z' " warning in the terminal. 
[image]
[image]
However, when I disabled the vio, the point cloud map works very well.

I wonder if there are any possible changes I should make in the code to fix this issue. Any suggestions are appreciated! 

Thank you

closed 2022-07-13 by stale[bot]

seajayshore, 2022-08-13:
How did you disable the VIO?

WillBao33, 2022-08-13:
> How did you disable the VIO?

Hi, 

I just simply not fed the image topic to r3live, so basically turned it back to fast-lio 2.

farhad-dalirani, 2023-01-20:
Hi,

How did you adopt the code for Velodyne-16?
Where did you find the handler and how did you change it?
Would you mind sending the code?

WillBao33, 2023-01-20:
> Hi,
> 
> How did you adopt the code for Velodyne-16? Where did you find the handler and how did you change it? Would you mind sending the code?

I used the one from R2LIVE.

farhad-dalirani, 2023-02-09:
> > Hi,
> > How did you adopt the code for Velodyne-16? Where did you find the handler and how did you change it? Would you mind sending the code?
> 
> I used the one from R2LIVE.

Hi @WillBao33 ,

I generate colored map with R3Live with Velodyne and a camera (RealSense). However, a significant drift happens.
I explained my problem with detail in this issue:
https://github.com/hku-mars/r3live/issues/157

If you look at it, it would be great.

Thanks,