hku-mars/r3live #110: I am trying to run the code as per the prerequisites and instructions but getting error when I am using the lauch command. THe error pops out.

Rakshit-bit, 2022-07-10:
cannot launch node of type [r3live/r3live_LiDAR_front_end]: Cannot locate node of type [r3live_LiDAR_front_end] in package [r3live]. Make sure file exists in package path and permission is set to executable (chmod +x)
ERROR: cannot launch node of type [r3live/r3live_mapping]: Cannot locate node of type [r3live_mapping] in package [r3live]. Make sure file exists in package path and permission is set to executable (chmod +x)

Rakshit-bit, 2022-07-10:
Also rviz is opening showing no -image error and Global warn: no TF data. Could you please provide any help? It will great. Thank you.
[image]

xutianq999, 2022-07-19:
I had the same problem as you and tormented me for a long time, the good news is that I have solved it. You can find the function packages with errors in the path "~r3live_ws/devel/lib/r3live". If you don't know where you put them, you can use the "locate" tool to find them. Then use "chmod +x" to add permissions to them. If you want to run with your own data set, there will be many questions later, I am also learning, if you want to communicate, 734609080@qq.com is my email.

closed 2022-08-03 by stale[bot]