hku-mars/r3live #113: ouster如何开启vio使其上色？

wanghui-min, 2022-07-27:
我使用提供的livox bag实现了建模及上色，发现ouster bag并不可以，目前我所采用的是ouster 64线雷达，实现了建模，但是并没有上色，已经修改了相机参数，请问该如何实现上色？

closed 2022-08-10 by wanghui-min

xuanxuanxuanhui, 2022-08-11:
您好，我想请问一下您解决这个问题了嘛？要如何开启vio子系统呀？

wanghui-min, 2022-08-11:
> 您好，我想请问一下您解决这个问题了嘛？要如何开启vio子系统呀？

解决了，我是先通过读取雷达imu的时间戳，赋值给视频流，然后进行发布到对应的话题，再将lunch文件下的ouster配置文件图片话题修改成/camera/image_color，配准什么的没做，配准效果很差，使用默认的矩阵是可以上色的
import argparse
import time

import numpy as np

import cv2
import rospy
from sensor_msgs.msg import Imu
from sensor_msgs.msg import CompressedImage

from cv_bridge import CvBridge, CvBridgeError
now_time=0
bridge = CvBridge()
def callback(data):
    global now_time
    now_time=data.header.stamp
    #rospy.loginfo(rospy.get_caller_id() + 'imu %s', now_time)
def main():
    """Publish a video as ROS messages.
    """
    # Patse arguments.
    # Set up node.
    rospy.init_node("video_publisher", anonymous=True)
    img_pub = rospy.Publisher("/camera/image_color/compressed", CompressedImage,
                              queue_size=2)

    # Open video.
    rospy.Subscriber('/os_cloud_node/imu', Imu, callback)
    video = cv2.VideoCapture(2)
    video.set(6, cv2.VideoWriter_fourcc('M', 'J', 'P', 'G'))
    video.set(5,30)
    video.set(3, 1280)
    video.set(4, 720)
    rate = rospy.Rate(10)
    msg = CompressedImage()
    global now_time
    while not rospy.is_shutdown() and video.grab():
        tmp, img = video.retrieve()
        if not tmp:
            print "Could not grab frame."
            break
        # Publish image.
        msg.header.stamp =now_time
        msg.format = "jpeg"
        msg.data = np.array(cv2.imencode('.jpg', img)[1]).tostring()
        msg.header.frame_id = str("world")
        img_pub.publish(msg)
        rate.sleep()

    return

if __name__ == '__main__':
    try:
        main()
    except rospy.ROSInterruptException:
        pass

Russell-fan, 2022-11-28:
> 

您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？

wanghui-min, 2022-11-28:
> > 
> 
> 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？

解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。

Russell-fan, 2022-11-28:
> > > 
> > 
> > 
> > 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？
> 
> 解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。

非常感谢您的回复，我刚刚接触这方面的内容，请问您方便开源一下您修改后的代码吗？或者可以麻烦您发邮件到1873006476@qq.com吗？我想学习一下，再次感谢您。

wanghui-min, 2022-11-28:
> > > > 
> > > 
> > > 
> > > 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？
> > 
> > 
> > 解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。
> 
> 非常感谢您的回复，我刚刚接触这方面的内容，请问您方便开源一下您修改后的代码吗？或者可以麻烦您发邮件到[1873006476@qq.com](mailto:1873006476@qq.com)吗？我想学习一下，再次感谢您。

因为在另一台计算机里头，等明天吧，你明天记得问我，我给你找找

Russell-fan, 2022-11-28:
> > > > > 
> > > > 
> > > > 
> > > > 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？
> > > 
> > > 
> > > 解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。
> > 
> > 
> > 非常感谢您的回复，我刚刚接触这方面的内容，请问您方便开源一下您修改后的代码吗？或者可以麻烦您发邮件到[1873006476@qq.com](mailto:1873006476@qq.com)吗？我想学习一下，再次感谢您。
> 
> 因为在另一台计算机里头，等明天吧，你明天记得问我，我给你找找

嗯嗯，非常感谢！

Russell-fan, 2022-11-29:
> > > > > 
> > > > 
> > > > 
> > > > 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？
> > > 
> > > 
> > > 解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。
> > 
> > 
> > 非常感谢您的回复，我刚刚接触这方面的内容，请问您方便开源一下您修改后的代码吗？或者可以麻烦您发邮件到[1873006476@qq.com](mailto:1873006476@qq.com)吗？我想学习一下，再次感谢您。
> 
> 因为在另一台计算机里头，等明天吧，你明天记得问我，我给你找找

上午好，请问您现在方便嘛？我的邮箱：1873006476@qq.com  麻烦您了

Little-cat-like-fish, 2022-12-15:
您好，我也是刚开始接触r3live相关内容，能不能麻烦您把修改后的代码也发给我一份，万分感谢！我的邮箱：852839352@qq.com非常感谢，麻烦您了！

wanghui-min, 2022-12-21:
> 305044611@qq.ocm

下次记得邮箱写对

zwjgit, 2022-12-21:
> > [305044611@qq.ocm](mailto:305044611@qq.ocm)
> 
> 下次记得邮箱写对

okk

Louis-Lv-Neu, 2023-11-28:
> > > 
> > 
> > 
> > 您好，请问一下您解决这个问题了吗？方便分享一下您是怎样做的吗？
> 
> 解决了，我是直接通过python2 ros获取到那个ouster点云数据的时间或者imu数据的时间，然后用opencv获取摄像头的图像，最后将RGB图像发送到对应的的话题就可以了，主要是时间同步了就可以。

您好，您方便给我也分享一份吗739373778@qq.com 谢谢您

CloudAlice, 2023-12-09:
您好，您方便给我也分享一份吗 1132593638@qq.com 谢谢您

jingyilon, 2024-03-04:
您好，我刚开始接触这些，如果您方便的话，能给我也分享一份吗（2427375498@qq.com） 不甚感激！

Hohaizzz, 2024-05-29:
您好，打扰您一下，最近也遇到了这个问题，能分享一下吗（1732808062@qq.com）特别感谢！

komorebi0315, 2024-06-28:
您好，可以分享一份吗（2314272795@qq.com）非常感谢！！！

kkaerma, 2025-04-09:
您好，您方便给我也分享一份吗（2778087067@qq.com） 谢谢您!!!!!!!