hku-mars/r3live #114: 如何添加外部imu

gongyue666, 2022-08-11:
你好，我利用海康相机+livox mid 70+ch108构建了手持设备，lidar和imu，lidar和camera都进行分别进行了标定。但是，在配置时，我发现config.yaml中只有camera和lidar的外参配置项目，没有和imu相关的外参配置方法。请问如何将imu外参加入？？
已经做过的尝试：修改common_lib.h中的第61行: static const Eigen::Vector3d Lidar_offeset_to_IMU, 但貌似这里只有平移，没有旋转相关的变量。

BaoZehan, 2022-08-12:
您好，您找到解决方法了么？我也准备用mid 70 +ch110

seajayshore, 2022-08-12:
I am trying to solve this problem too.

In [this issue](https://github.com/hku-mars/r3live/issues/14) user @tlin442 has discribed some detail of how they solved the problem.

They have a fork [here](https://github.com/tlin442/r3live) which has some changes made to `LiDAR_front_end.cpp` to allow for rotational extrinsics + changes to `common_lib.h` (as you noticed) for translational extrinsic. Also don't forget to modify the calibration YAML with the added extrinsics.

(Please note that tlin442 also made changes to allow for fisheye cameras so some changes in other files may not actully work for your data - you may need to manually review the code).

I am currently using tlin442's changes as a starting point to improve external IMU for my own version but with other modifications for other reasons. 

Good luck!

gongyue666, 2022-08-15:
Thx so much， I will try it