hku-mars/r3live #121: catkin_ws 时报错，请问如何解决，期待您的回复，谢谢

abnerlam, 2022-09-20:
Base path: /home/nuc/catkin_ws
Source space: /home/nuc/catkin_ws/src
Build space: /home/nuc/catkin_ws/build
Devel space: /home/nuc/catkin_ws/devel
Install space: /home/nuc/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/nuc/catkin_ws/build"
####
####
#### Running command: "make -j8 -l8" in "/home/nuc/catkin_ws/build"
####
[  0%] Built target geometry_msgs_generate_messages_cpp
[  0%] Built target geometry_msgs_generate_messages_nodejs
[  0%] Built target geometry_msgs_generate_messages_eus
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_py
[  6%] Built target test_timer
[  6%] Built target r3live_generate_messages_cpp
[  6%] Built target r3live_generate_messages_nodejs
[  9%] Built target r3live_generate_messages_eus
[  9%] Built target r3live_generate_messages_lisp
[ 15%] Built target r3live_LiDAR_front_end
[ 15%] Built target r3live_generate_messages_py
[ 18%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o
[ 21%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o
[ 24%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o
[ 24%] Built target r3live_generate_messages
[ 27%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o
[ 33%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o
[ 33%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o
[ 36%] Building CXX object r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o
/home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp: In member function ‘int LK_optical_flow_kernel::track_image(const cv::Mat&, const std::vector<cv::Point_<float> >&, std::vector<cv::Point_<float> >&, std::vector<unsigned char>&, int)’:
/home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp:803:10: error: invalid initialization of reference of type ‘const cv::ParallelLoopBody&’ from expression of type ‘LK_optical_flow_kernel::track_image(const cv::Mat&, const std::vector<cv::Point_<float> >&, std::vector<cv::Point_<float> >&, std::vector<unsigned char>&, int)::<lambda(const cv::Range&)>’
         });
          ^
In file included from /usr/include/opencv2/core.hpp:3216:0,
                 from /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.hpp:3,
                 from /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp:45:
/usr/include/opencv2/core/utility.hpp:478:17: note: in passing argument 2 of ‘void cv::parallel_for_(const cv::Range&, const cv::ParallelLoopBody&, double)’
 CV_EXPORTS void parallel_for_(const Range& range, const ParallelLoopBody& body, double nstripes=-1.);
                 ^~~~~~~~~~~~~
r3live/CMakeFiles/r3live_mapping.dir/build.make:230: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o] Error 1
make[2]: *** 正在等待未完成的任务....
^CTraceback (most recent call last):
  File "/opt/ros/melodic/bin/catkin_make", line 306, in <module>
    sys.exit(main())
  File "/opt/ros/melodic/bin/catkin_make", line 249, in main
    r3live/CMakeFiles/r3live_meshing.dir/build.make:110: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o] 中断
r3live/CMakeFiles/r3live_mapping.dir/build.make:302: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o] 中断
r3live/CMakeFiles/r3live_mapping.dir/build.make:86: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o] 中断
r3live/CMakeFiles/r3live_mapping.dir/build.make:206: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o] 中断
r3live/CMakeFiles/r3live_mapping.dir/build.make:158: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o] 中断
r3live/CMakeFiles/r3live_mapping.dir/build.make:62: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o] 中断
run_command(cmd, make_path)
  File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/builder.py", line 241, in run_command
CMakeFiles/Makefile2:2731: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/all' failed
make[1]: *** [r3live/CMakeFiles/r3live_mapping.dir/all] 中断
CMakeFiles/Makefile2:536: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/all' failed
make[1]: *** [r3live/CMakeFiles/r3live_meshing.dir/all] 中断
Makefile:140: recipe for target 'all' failed
make: *** [all] 中断
    proc.wait()
  File "/usr/lib/python2.7/subprocess.py", line 1099, in wait
[... 74 lines cut ...]
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
r3live/CMakeFiles/r3live_meshing.dir/build.make:110: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o] Error 1
CMakeFiles/Makefile2:536: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/all' failed
make[1]: *** [r3live/CMakeFiles/r3live_meshing.dir/all] Error 2
make[1]: *** 正在等待未完成的任务....
/home/nuc/catkin_ws/src/r3live/src/rgb_map/pointcloud_rgbd.cpp: In member function ‘void Global_map::save_and_display_pointcloud(std::__cxx11::string, std::__cxx11::string, int)’:
/home/nuc/catkin_ws/src/r3live/src/rgb_map/pointcloud_rgbd.cpp:661:11: warning: ignoring return value of ‘int system(const char*)’, declared with attribute warn_unused_result [-Wunused-result]
     system(std::string("pcl_viewer ").append(dir_name).append("/rgb_pt.pcd").c_str());
     ~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
r3live/CMakeFiles/r3live_mapping.dir/build.make:302: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed
make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o] Error 1
/home/nuc/catkin_ws/src/r3live/src/r3live_vio.cpp: In member function ‘void R3LIVE::load_vio_parameters()’:
/home/nuc/catkin_ws/src/r3live/src/r3live_vio.cpp:452:68: warning: format ‘%d’ expects argument of type ‘int’, but argument 3 has type ‘std::vector<double, std::allocator<double> >::size_type {aka long unsigned int}’ [-Wformat=]
         printf( "Load camera data size = %d, %d, %d, %d\n", ( int ) camera_intrinsic_data.size(), camera_dist_coeffs_data.size(),
                                                                                                   ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                 camera_ext_R_data.size(), camera_ext_t_data.size() );
                                                                    ^
/home/nuc/catkin_ws/src/r3live/src/r3live_vio.cpp:452:68: warning: format ‘%d’ expects argument of type ‘int’, but argument 4 has type ‘std::vector<double, std::allocator<double> >::size_type {aka long unsigned int}’ [-Wformat=]
/home/nuc/catkin_ws/src/r3live/src/r3live_vio.cpp:452:68: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘std::vector<double, std::allocator<double> >::size_type {aka long unsigned int}’ [-Wformat=]
CMakeFiles/Makefile2:2731: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/all' failed
make[1]: *** [r3live/CMakeFiles/r3live_mapping.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

closed 2022-09-28 by abnerlam

Ricardo-Meitang, 2023-06-13:
我也不会

melodydeath, 2023-11-08:
> Base path: /home/nuc/catkin_ws Source space: /home/nuc/catkin_ws/src Build space: /home/nuc/catkin_ws/build Devel space: /home/nuc/catkin_ws/devel Install space: /home/nuc/catkin_ws/install
> 
> #### Running command: "make cmake_check_build_system" in "/home/nuc/catkin_ws/build"
> #### Running command: "make -j8 -l8" in "/home/nuc/catkin_ws/build"
> [ 0%] Built target geometry_msgs_generate_messages_cpp [ 0%] Built target geometry_msgs_generate_messages_nodejs [ 0%] Built target geometry_msgs_generate_messages_eus [ 0%] Built target geometry_msgs_generate_messages_lisp [ 0%] Built target geometry_msgs_generate_messages_py [ 6%] Built target test_timer [ 6%] Built target r3live_generate_messages_cpp [ 6%] Built target r3live_generate_messages_nodejs [ 9%] Built target r3live_generate_messages_eus [ 9%] Built target r3live_generate_messages_lisp [ 15%] Built target r3live_LiDAR_front_end [ 15%] Built target r3live_generate_messages_py [ 18%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o [ 21%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o [ 24%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o [ 24%] Built target r3live_generate_messages [ 27%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o [ 33%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o [ 33%] Building CXX object r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o [ 36%] Building CXX object r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp: In member function ‘int LK_optical_flow_kernel::track_image(const cv::Mat&, const std::vector<cv::Point_ >&, std::vector<cv::Point_ >&, std::vector&, int)’: /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp:803:10: error: invalid initialization of reference of type ‘const cv::ParallelLoopBody&’ from expression of type ‘LK_optical_flow_kernel::track_image(const cv::Mat&, const std::vector<cv::Point_ >&, std::vector<cv::Point_ >&, std::vector&, int)::<lambda(const cv::Range&)>’ }); ^ In file included from /usr/include/opencv2/core.hpp:3216:0, from /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.hpp:3, from /home/nuc/catkin_ws/src/r3live/src/optical_flow/lkpyramid.cpp:45: /usr/include/opencv2/core/utility.hpp:478:17: note: in passing argument 2 of ‘void cv::parallel_for_(const cv::Range&, const cv::ParallelLoopBody&, double)’ CV_EXPORTS void parallel_for_(const Range& range, const ParallelLoopBody& body, double nstripes=-1.); ^~~~~~~~~~~~~ r3live/CMakeFiles/r3live_mapping.dir/build.make:230: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/optical_flow/lkpyramid.cpp.o] Error 1 make[2]: *** 正在等待未完成的任务.... ^CTraceback (most recent call last): File "/opt/ros/melodic/bin/catkin_make", line 306, in sys.exit(main()) File "/opt/ros/melodic/bin/catkin_make", line 249, in main r3live/CMakeFiles/r3live_meshing.dir/build.make:110: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_meshing.dir/src/rgb_map/pointcloud_rgbd.cpp.o] 中断 r3live/CMakeFiles/r3live_mapping.dir/build.make:302: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/rgb_map/pointcloud_rgbd.cpp.o] 中断 r3live/CMakeFiles/r3live_mapping.dir/build.make:86: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live_lio.cpp.o] 中断 r3live/CMakeFiles/r3live_mapping.dir/build.make:206: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live_vio.cpp.o] 中断 r3live/CMakeFiles/r3live_mapping.dir/build.make:158: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/loam/IMU_Processing.cpp.o] 中断 r3live/CMakeFiles/r3live_mapping.dir/build.make:62: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o' failed make[2]: *** [r3live/CMakeFiles/r3live_mapping.dir/src/r3live.cpp.o] 中断 run_command(cmd, make_path) File "/opt/ros/melodic/lib/python2.7/dist-packages/catkin/builder.py", line 241, in run_command CMakeFiles/Makefile2:2731: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/all' failed make[1]: *** [r3live/CMakeFiles/r3live_mapping.dir/all] 中断 CMakeFiles/Makefile2:536: recipe for target 'r3live/CMakeFiles/r3live_meshing.dir/all' failed make[1]: *** [r3live/CMakeFiles/r3live_meshing.dir/all] 中断 Makefile:140: recipe for target 'all' failed make: *** [all] 中断 proc.wait() File "/usr/lib/python2.7/subprocess.py", line 1099, in wait pid, sts = _eintr_retry_call(os.waitpid, self.pid, 0) File "/usr/lib/python2.7/subprocess.py", line 125, in _eintr_retry_call return func(*args) KeyboardInterrupt nuc@nuc-algorithm:~/catkin_ws$ catkin_make Base path: /home/nuc/catkin_ws Source space: /home/nuc/catkin_ws/src Build space: /home/nuc/catkin_ws/build Devel space: /home/nuc/catkin_ws/devel Install space: /home/nuc/catkin_ws/install
> 
[... cut ...] /src/r3live/src/r3live_vio.cpp:452:68: warning: format ‘%d’ expects argument of type ‘int’, but argument 5 has type ‘std::vector<double, std::allocator >::size_type {aka long unsigned int}’ [-Wformat=] CMakeFiles/Makefile2:2731: recipe for target 'r3live/CMakeFiles/r3live_mapping.dir/all' failed make[1]: *** [r3live/CMakeFiles/r3live_mapping.dir/all] Error 2 Makefile:140: recipe for target 'all' failed make: *** [all] Error 2 Invoking "make -j8 -l8" failed

Hi, were you able to solve this issue?

abnerlam, 2023-11-08:
> Hi, were you able to solve this issue?

Yes, check your python and ROS opencv version.

melodydeath, 2023-11-08:
> Yes, check your python and ROS opencv version.

Ok, I've solved it, thank you