hku-mars/r3live #130: 我用了自己数据集，在r2live上能够正常运行，可是在r3live不能运行（已经用livox_camera_calib标定过了）

liefengguo, 2022-10-23:
就是显示在VIO里 Camera-frame一直是2，感觉像是vio的处理出了问题
用的设备是海康工业相机和avia，我也怀疑是不是我这边海康ros提供数据环境有问题。我也是在git找的海康ros。
现在状态是lio系统能正常运行，vio运行异常，track_img 也没有生成出来。

525753936, 2022-11-02:
您好，请问您是怎么解决vio部分的问题的，我的问题跟您的很像，lio部分正常，vio能显示track image 但无法生成彩色点云，能否分享一下您修改的配置文件部分，还有您解决这个问题的方法经验

525753936, 2022-11-02:
> 

[image]

yipengzhu0809, 2022-11-02:
您好，我和你相机、雷达的config都完全一致，也遇到了同样的问题，想请教一下是怎么解决的，谢谢。

closed 2022-11-18 by stale[bot]

zhuoyuan00, 2023-01-03:
> 您好，请问您是怎么解决vio部分的问题的，我的问题跟您的很像，lio部分正常，vio能显示track image 但无法生成彩色点云，能否分享一下您修改的配置文件部分，还有您解决这个问题的方法经验

问题相同   请问解决了吗

liefengguo, 2023-01-04:
> 

我当时是看csdn的一篇博客才解决的，你可以去看看https://blog.csdn.net/Ken_liefeng/article/details/127502634，你就跟着他的走就可以了

liefengguo, 2023-01-04:
> > 您好，请问您是怎么解决vio部分的问题的，我的问题跟您的很像，lio部分正常，vio能显示track image 但无法生成彩色点云，能否分享一下您修改的配置文件部分，还有您解决这个问题的方法经验
> 
> 问题相同 请问解决了吗

我当时是看csdn的一篇博客才解决的，你可以去看看[https://blog.csdn.net/Ken_liefeng/article/details/127502634，你就跟着他的走就可以了](https://blog.csdn.net/Ken_liefeng/article/details/127502634%EF%BC%8C%E4%BD%A0%E5%B0%B1%E8%B7%9F%E7%9D%80%E4%BB%96%E7%9A%84%E8%B5%B0%E5%B0%B1%E5%8F%AF%E4%BB%A5%E4%BA%86)

liefengguo, 2023-01-04:
> 您好，我和你相机、雷达的config都完全一致，也遇到了同样的问题，想请教一下是怎么解决的，谢谢。

我当时是看csdn的一篇博客才解决的，你可以去看看[https://blog.csdn.net/Ken_liefeng/article/details/127502634，你就跟着他的走就可以了](https://blog.csdn.net/Ken_liefeng/article/details/127502634%EF%BC%8C%E4%BD%A0%E5%B0%B1%E8%B7%9F%E7%9D%80%E4%BB%96%E7%9A%84%E8%B5%B0%E5%B0%B1%E5%8F%AF%E4%BB%A5%E4%BA%86)

liefengguo, 2023-01-04:
> 就是显示在VIO里 Camera-frame一直是2，感觉像是vio的处理出了问题 用的设备是海康工业相机和avia，我也怀疑是不是我这边海康ros提供数据环境有问题。我也是在git找的海康ros。 现在状态是lio系统能正常运行，vio运行异常，track_img 也没有生成出来。 [image] [image] 已解决

我当时是看csdn的一篇博客才解决的，你可以去看看[https://blog.csdn.net/Ken_liefeng/article/details/127502634]，你就跟着他的走就可以了(https://blog.csdn.net/Ken_liefeng/article/details/127502634%EF%BC%8C%E4%BD%A0%E5%B0%B1%E8%B7%9F%E7%9D%80%E4%BB%96%E7%9A%84%E8%B5%B0%E5%B0%B1%E5%8F%AF%E4%BB%A5%E4%BA%86)