hku-mars/r3live #137: How to use velodyne16 +  camera

Fafa-DL, 2022-11-23:
Thank you very much for your work. I have updated the velodyne handler and can complete the map building, but the camera cannot be processed. The terminal clearly received the camera-frame and the number is constantly updated.

The problem lies in [r3live_vio.cpp](https://github.com/hku-mars/r3live/blob/master/r3live/src/r3live_vio.cpp#L1166)

But my bag has already done time synchronization, I don’t know how to solve the problem, thank you.

closed 2022-11-23 by Fafa-DL

farhad-dalirani, 2023-01-18:
Could you resolve the problem?
Velodyne 16 does not have internal IMU, how did you calibrated it with LiDAR?

farhad-dalirani, 2023-02-09:
> Could you resolve the problem? Velodyne 16 does not have internal IMU, how did you calibrated it with LiDAR?

I put an IMU under Velodyne, I tried my best to align their axes. When I use R3Live with just LiDAR, it works fine.