hku-mars/r3live #139: 有关地图大小在增长，但其中没有任何点的问题

Orwlit, 2022-11-26:
您好！首先非常感谢您将效果如此出色的R3LIVE代码开源，但我在用自己的设备（realsense D435i + livox AVIA）测试时遇到了两个问题：整个过程中里程计是正常的，相机和设备间的外参也经过标定，但地图却没有任何保留下来的数据，同时记录系统状态的标题也被一次次重复打印。

如下图所示，随着相机帧和雷达帧的不断输入，可以看到Pts in maps在一直增长，但rviz中始终没有任何显示出的地图点，同时我发现发布rgb点云的话题中没有任何数据，但数组的长度却一直在增加。

另外很奇怪的一点是，如图左下角终端内容中，记录各种系统信息的数字在持续更新，但其标题却跟着数字被一次次打印出来，我不知道这个问题会不会和地图没有任何内容有关。

[image]

另附上运行时终端打印的所有信息：
```mini@x86pc:~/myCode/test_ws $ roslaunch r3live r3live_bag.launch         
... logging to /home/mini/.ros/log/9a692f38-6d56-11ed-aa3e-e1141e553ca6/roslaunch-x86pc-8724.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://x86pc:34931/

SUMMARY
========

PARAMETERS
 * /IMU_topic: /livox/imu
 * /Image_topic: /camera/color/ima...
 * /LiDAR_pointcloud_topic: /laser_cloud_flat
 * /Lidar_front_end/N_SCANS: 6
 * /Lidar_front_end/lidar_type: 1
 * /Lidar_front_end/point_step: 1
 * /Lidar_front_end/using_raw_point: 1
 * /r3live_common/append_global_map_point_step: 4
 * /r3live_common/estimate_i2c_extrinsic: 1
 * /r3live_common/estimate_intrinsic: 1
 * /r3live_common/if_dump_log: 0
 * /r3live_common/image_downsample_ratio: 1
 * /r3live_common/map_output_dir: /home/mini/r3live...
 * /r3live_common/maximum_vio_tracked_pts: 600
 * /r3live_common/minimum_pts_size: 0.01
 * /r3live_common/pub_pt_minimum_views: 3
 * /r3live_common/record_offline_map: 1
 * /r3live_lio/filter_size_corner: 0.3
 * /r3live_lio/filter_size_map: 0.3
 * /r3live_lio/filter_size_surf: 0.3
 * /r3live_lio/filter_size_surf_z: 0.3
 * /r3live_lio/lidar_time_delay: 0
 * /r3live_lio/lio_update_point_step: 4
 * /r3live_lio/max_iteration: 2
 * /r3live_vio/camera_dist_coeffs: [-0.0944205499243...
 * /r3live_vio/camera_ext_R: [-0.0120759, -0.9...
 * /r3live_vio/camera_ext_t: [0.0385914, 0.000...
 * /r3live_vio/camera_intrinsic: [605.114868164062...
 * /r3live_vio/image_height: 480
 * /r3live_vio/image_width: 640
 * /rosdistro: noetic
 * /rosversion: 1.15.14

NODES
  /
    r3live_LiDAR_front_end (r3live/r3live_LiDAR_front_end)
    r3live_mapping (r3live/r3live_mapping)
    rvizvisualisation (rviz/rviz)

ROS_MASTER_URI=http://localhost:11311

process[r3live_LiDAR_front_end-1]: started with pid [8745]
process[r3live_mapping-2]: started with pid [8746]
HORIZON
process[rvizvisualisation-3]: started with pid [8752]
=============================================================
App name   : R3LIVE: A Robust, Real-time, RGB-colored, LiDAR-Inertial-Visual tightly-coupled state Estimation and mapping package
Build date : Nov 25 2022  18:26:01
CPU infos  : 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
RAM infos  : 31.10GB Physical Memory 30.52GB Virtual Memory
OS  infos  : Linux 5.15.0-53-generic (x86_64)
Home dir   : /home/mini/
Current dir: /home/mini/.ros/
Date mow   : 2022/11/26 14:50:55
=============================================================
+++++++++++++++++++++++++++++++++++++++++++++++
Here is the your software environments: 
GCC version          : 9.4.0
Boost version        : 1.71.0
Eigen version        : 3.3.7
OpenCV version       : 4.2.0
+++++++++++++++++++++++++++++++++++++++++++++++
Multi thread started 
[Ros_parameter]: /LiDAR_pointcloud_topic ==> /laser_cloud_flat
[Ros_parameter]: /IMU_topic ==> /livox/imu
[Ros_parameter]: /Image_topic ==> /camera/color/image_raw
======= Summary of subscribed topics =======
LiDAR pointcloud topic: /laser_cloud_flat
IMU topic: /livox/imu
Image topic: /camera/color/image_raw
Image compressed topic: /camera/color/image_raw
=======        -End-                =======
[Ros_parameter]: r3live_common/map_output_dir ==> /home/mini/r3live_output
[Ros_parameter]: r3live_common/append_global_map_point_step ==> 4
[Ros_parameter]: r3live_common/recent_visited_voxel_activated_time ==> 0
[Ros_parameter]: r3live_common/maximum_image_buffer ==> 20000
[Ros_parameter]: r3live_common/tracker_minimum_depth ==> 0.1
[Ros_parameter]: r3live_common/tracker_maximum_depth ==> 200
[Ros_parameter]: r3live_common/track_windows_size ==> 40
[Ros_parameter]: r3live_common/minimum_pts_size ==> 0.01
[Ros_parameter]: r3live_common/record_offline_map ==> 1
[Ros_parameter]: r3live_common/pub_pt_minimum_views ==> 3
[Ros_parameter]: r3live_common/image_downsample_ratio ==> 1
[Ros_parameter]: r3live_common/esikf_iter_times ==> 2
[Ros_parameter]: r3live_common/estimate_i2c_extrinsic ==> 1
[Ros_parameter]: r3live_common/estimate_intrinsic ==> 1
[Ros_parameter]: r3live_common/maximum_vio_tracked_pts ==> 600
[Ros_parameter]: r3live_lio/dense_map_enable ==> 1
[Ros_parameter]: r3live_lio/lidar_time_delay ==> 0
[Ros_parameter]: r3live_lio/max_iteration ==> 2
[Ros_parameter]: r3live_lio/fov_degree ==> 360
[Ros_parameter]: r3live_lio/voxel_downsample_size_surf ==> 0.3
[Ros_parameter]: r3live_lio/voxel_downsample_size_axis_z ==> 0.3
[Ros_parameter]: r3live_lio/filter_size_map ==> 0.3
[Ros_parameter]: r3live_lio/cube_side_length ==> 1e+07
[Ros_parameter]: r3live_lio/maximum_pt_kdtree_dis ==> 0.5
[Ros_parameter]: r3live_lio/maximum_res_dis ==> 0.3
[Ros_parameter]: r3live_lio/planar_check_dis ==> 0.1
[Ros_parameter]: r3live_lio/long_rang_pt_dis ==> 500
[Ros_parameter]: r3live_lio/publish_feature_map ==> 0
[Ros_parameter]: r3live_lio/lio_update_point_step ==> 4
[Ros_parameter]: r3live_vio/Camera Intrinsic: 
605.115       0  326.93
      0 605.208 249.693
      0       0       1
[Ros_parameter]: r3live_vio/Camera distcoeff:  -0.0944205   0.0946728 -0.00807971 8.07461e-05     -0.0222
[Ros_parameter]: r3live_vio/Camera extrinsic R: 
-0.0120759  -0.999497  0.0293101
-0.0124686 -0.0291595  -0.999497
  0.999849 -0.0124353 -0.0121102
[Ros_parameter]: r3live_vio/Camera extrinsic T:   0.0385914 0.000396904   -0.028381
Thread_id [140613616023296] set CPU-1 affinity success.
Thread_id [140597538277120] set CPU-2 affinity success.
Thread_id [140597521491712] set CPU-5 affinity success.
Thread_id [140613624416000] set CPU-4 affinity success.
Thread_id [140613632808704] set CPU-0 affinity success.
Thread_id [140597529884416] set CPU-3 affinity success.
[ INFO] [1669445456.704610243]: IMU Initializing: 5.0 %
[ WARN] [1669445456.705452658]: Reset ImuProcess
[ INFO] [1669445456.705488650]: IMU Initials: Gravity: 0.0000 0.0000 9.8050; state.bias_g: 0.0037 -0.0024 0.0003; acc covarience: 0.10000000 0.10000000 0.10000000; gry covarience: 0.10000000 0.10000000 0.10000000
not ready for odometry

======================= R3LIVE Dashboard ======================
| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    8        |    0         | 25580        |    277          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    8        |    25        | 30315        |    287          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    8        |    26        | 30315        |    288          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    9        |    28        | 30315        |    290          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    9        |    28        | 35100        |    290          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    9        |    29        | 35100        |    290          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    10       |    31        | 35100        |    295          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    10       |    31        | 39677        |    296          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:56  |    10       |    32        | 39677        |    298          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:57  |    11       |    34        | 39677        |    304          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:57  |    11       |    34        | 44391        |    304          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:57  |    11       |    35        | 44391        |    306          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |
|   14:50:57  |    12       |    37        | 44391        |    307          ****| System-time | LiDAR-frame | Camera-frame |  Pts in maps | Memory used (Mb) |

closed 2022-12-11 by stale[bot]

komorebi0315, 2023-03-01:
你好，请问问题解决了吗？我也遇到了这个问题

Orwlit, 2023-03-01:
解决了，是我的粗心大意导致了这个问题，lidar-cam的变换矩阵我填反了，需要求逆。
我用的是官方的lidar-cam标定：https://github.com/hku-mars/livox_camera_calib
r3live中需要把标定好的结果求逆。

Orwlit, 2023-03-01:
> 你好，请问问题解决了吗？我也遇到了这个问题

解决了，是我的粗心大意导致了这个问题，lidar-cam的变换矩阵我填反了，需要求逆。
我用的是官方的lidar-cam标定：https://github.com/hku-mars/livox_camera_calib
r3live中需要把标定好的结果求逆。

komorebi0315, 2023-03-01:
太感谢了，我也是这个问题，谢谢您！！！

YangZhouDL, 2024-01-19:
> > 你好，请问问题解决了吗？我也遇到了这个问题
> 
> 解决了，是我的粗心大意导致了这个问题，lidar-cam的变换矩阵我填反了，需要求逆。 我用的是官方的lidar-cam标定：https://github.com/hku-mars/livox_camera_calib r3live中需要把标定好的结果求逆。

你好，请问lidar-cam的变换矩阵，是lidar到camera的变换还是camera到lidar的变换呢

Orwlit, 2024-01-19:
> > > 你好，请问问题解决了吗？我也遇到了这个问题
> 
> > 
> 
> > 解决了，是我的粗心大意导致了这个问题，lidar-cam的变换矩阵我填反了，需要求逆。 我用的是官方的lidar-cam标定：https://github.com/hku-mars/livox_camera_calib r3live中需要把标定好的结果求逆。
> 
> 
> 
> 你好，请问lidar-cam的变换矩阵，是lidar到camera的变换还是camera到lidar的变换呢

我也记不清了，你要不都试试呗