hku-mars/r3live #165: r3live外置imu发生漂移，fast-lio效果较好

fanshixiong, 2023-03-04:
你好，非常感谢你们的工作，我在使用r3live时出现了很大的漂移，尤其是转弯处，直线处效果还行，但是同样的参数配置在fast-lio2里却取得了很好的效果。
我的r3live版本是添加了外置imu旋转的版本，仓库在这里：@tiny442 https://github.com/tlin442/r3live
我使用的激光雷达是livox mid-70，外置imu，使用的是mynt-D1000相机和imu，同时使用了修改过后的livox driver。
我的电脑是ubuntu20.04。
这是我的配置文件：
```
Lidar_front_end:
   lidar_type: 1   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 4
   using_raw_point: 1
   point_step: 1
   min_intensity: 20.0
   min_dist: 0.1
   lidar_imu_rotm:
      # LiDAR is mounted rotated by 90 deg
      #[1, 0, 0,
      # 0, 0, 1,
      # 0, -1, 0]
      [ 0.020712, -0.999783,  0.002253,
   0.073848,  0.003778,  0.997262,
  -0.997054, -0.020489,  0.073910]
   lidar_imu_tranm: 
      [0.037628, 0.067744, 0.098026]

r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 20  # The point step of append point to global map. [default = 4]

r3live_vio:
   
   # This is the real camera intrinsics used for computing the undistort map for fisheye
   camera_intrinsic_real:
      [655.005, 0, 679.029, 
      0, 656.097, 358.596, 
      0, 0, 1 ] 
   # This is the projection intrinsics used for the rest of the calculations
   camera_intrinsic:
      [655.005, 0, 679.029, 
      0, 656.097, 358.596, 
      0, 0, 1 ] 
      # [698.4 , 0, 650.593, 
      # 0, 698.4, 359.906, 
      # 0, 0, 1 ] 
   # Hacked in r3live_vio.cpp to be fisheye instead of normal pinhole
   camera_dist_coeffs: [-0.238605, 0.0435143, 0.000366211, -0.00272751, 0]  #k1, k2, p1, p2, k3
  # Camera->IMU value derived from T265 factory calibration.
   camera_ext_R:
         [0.886267, -0.0596407,  -0.459319,
         0.0143203, -0.987673,   0.155877,
         -0.462953, -0.144726,  -0.874488]
         #[0, 0, 1,
         # -1, 0, 0,
         # 0, -1, 0]
   #camera_ext_t: [-0.013776, 0.0347174, -0.0972432] 
   camera_ext_t: [ 0, 0, 0 ]
   
r3live_lio:        
   lio_update_point_step: 6   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: -0.092132        # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30
   publish_feature_map: 1
```
这是launch文件：
```
<launch>
    <!-- Subscribed topics -->
    <param name="LiDAR_pointcloud_topic" type="string" value="laser_cloud_flat" />
    <param name="IMU_topic" type="string" value= "/mynteye/imu/data_raw" />
    <param name="Image_topic" type="string" value= "/mynteye/right/image_color" />
    <param name="r3live_common/map_output_dir" type="string" value="$(env HOME)/r3live_output" />
    <rosparam command="load" file="$(find r3live)/../config/r3live_mid70_t265.yaml" />

    <node pkg="r3live" type="r3live_LiDAR_front_end" name="r3live_LiDAR_front_end"  output="screen" required="true"/>
    <node pkg="r3live" type="r3live_mapping" name="r3live_mapping" output="screen" required="true" />
    
    <arg name="rviz" default="1" />
    <group if="$(arg rviz)">
        <node name="rvizvisualisation" pkg="rviz" type="rviz" output="log" args="-d $(find r3live)/../config/rviz/r3live_rviz_config.rviz" />
    </group>
</launch>

```
请问有哪些地方有问题呢，谢谢回复，

closed 2023-03-20 by stale[bot]

fanshixiong, 2023-03-30:
imu内参标定用的imu_tools，相机和imu使用的是kalibr标定，重投影误差在1.5个像素左右。
在fast-lio2上测试结果较好，vins上测试结果一般，但是在r3lvie中却有着明显的漂移。
[image]
[image]
请问解决漂移有什么好的解决办法吗？谢谢答复。

jllk47, 2023-12-17:
> imu内参标定用的imu_tools，相机和imu使用的是kalibr标定，重投影误差在1.5个像素左右。 在fast-lio2上测试结果较好，vins上测试结果一般，但是在r3lvie中却有着明显的漂移。 [image] [image] 请问解决漂移有什么好的解决办法吗？谢谢答复。

您好想问您一下再您修改的r3live版本中针孔式相机可以使用嘛，如果不可以需要修改哪些地方呢