hku-mars/r3live #167: 可以用mid系列的雷达实现吗？比如mid-70

LiSi0205, 2023-03-08:
想知道把avia雷达换成mid-70可以复现出该算法的效果吗？

ziv-lin, 2023-03-16:
Yes, since the mid-70 do not have a built-in IMU yes,  you need to calibrate the spatial extrinsic and time-off between IMU and LiDAR. 
For the LiDAR-IMU calibration, please refer to this repo developed by our team: [**hku-mars/LiDAR_IMU_Init**](https://github.com/hku-mars/LiDAR_IMU_Init)

fanshixiong, 2023-04-01:
@LiSi0205 How did you solve the drifting problem? I have a drifting problem with Livox Lidar Mid-70 + MYNT IMU + MYNT camera.
I explained it in detail. It would be great if look at it:
https://github.com/hku-mars/r3live/issues/173
Thanks.

A-K11Xue, 2023-04-09:
Client [/r3live_mapping] wants topic /livox/lidar to have datatype/md5sum [sensor_msgs/PointCloud2/1158d486dd51d683ce2f1be655c3c181], but our version has [livox_ros_driver/CustomMsg/e4d6829bdfe657cb6c21a746c86b21a6]. Dropping connection.
How do I fix it? I used to modify the md5 code to be consistent, but it didn't solve the problem.
Thanks!

closed 2023-04-25 by stale[bot]