hku-mars/r3live #168: Calibration  between camera and lidar

farhad-dalirani, 2023-03-10:
Hello,

I use r3live with velodyne lidar. I calibrate camera and the lidar and in my calibration toolbox, the calibration result is good.

However, when I run R3Live, there is around 5 degree misalignment between the lidar and camera. For one case, I did not change sensors and I recorded a rosbag for on minutes. It was interesting for me that there is around 5 degree error in mapping between lidar and camera.

Do you know what can be the source of the problem?

In the right side of the below room there is a red cone on the ground, as you an see because of several degree misalignment, the red color are projected off the cone. Also, in the middle of the room, there is a diamond shape frame with yellow boundary. As it can be seen, some parts of the yellow boundary are projected on the wall.
[image]

farhad-dalirani, 2023-03-17:
The problem was my calibration tool. It was not accurate. A highly accurate lidar-camera calibration should be used.

closed 2023-03-17 by farhad-dalirani

esaumar, 2023-04-11:
Thanks @farhad-dalirani! One last question, have you tried calibrating the IMU/LiDAR vs Camera with this repo https://github.com/ethz-asl/kalibr ?

farhad-dalirani, 2023-04-11:
@esaumar no, I have not. I used Velodyne LiDAR which does not have an internal IMU, I used an external IMU; however, I put it under the LiDAR and tried my best to align the axes of the two sensors by my hand. It worked for me.