hku-mars/r3live #169: LiDAR incoming frame too old, need to be drop!!!

sreejithsreeni, 2023-03-16:
I have an issue with Hesai Lidar and D435i, r3live mapping. When I launch the mapping file I got the following error.
not ready for odometry
**LiDAR incoming frame too old, need to be drop!!!**

Please assist to solve the issue/

ziv-lin, 2023-03-16:
Your problem might be caused mainly by the misalignment of the timestamp. 

Since the LiDAR data and IMU data published by the official Livox-ros-driver is with the timestamp of LiDAR (started from 0 in each recording), and the timestamp of the image is usually recorded with the timestamp of the operation system. To make them working under the same time-based, we modified the source code of Livox-ros-driver, which is available at [here](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE). We suggest you replace the official driver with it when sampling your own data for R3LIVE.

[**Livox-ros-driver for R2/R3LIVE**](https://github.com/ziv-lin/livox_ros_driver_for_R2LIVE)

farhad-dalirani, 2023-03-17:
Also, I got this error, my problem was the use of loop in rosbab play.

sreesms, 2023-03-20:
> Your problem might be caused mainly by the misalignment of the timestamp.
>
> Since the LiDAR data and IMU data published by the official Livox-ros-driver is with the timestamp of LiDAR (started from 0 [...]

But I am not using Livox Lidar, I am using Hesai Lidar, Should I change the timestamp of Hesai driver file ?

sreesms, 2023-03-20:
> Also, I got this error, my problem was the use of loop in rosbab play.

I don't understand what you said, loop in rosbag? Could you please explain a bit?

sreesms, 2023-03-20:
> Your problem might be caused mainly by the misalignment of the timestamp.
>
> Since the LiDAR data and IMU data published by the official Livox-ros-driver is with the timestamp of LiDAR (started from 0 [...]

Yes, you were right, when I changed the timestamp to 'realtime' in Hesai driver file this error gone. Thank you for your help, appreciate it.

sreejithsreeni, 2023-03-20:
> Your problem might be caused mainly by the misalignment of the timestamp.
>
> Since the LiDAR data and IMU data published by the official Livox-ros-driver is with the timestamp of LiDAR (started from 0 [...]

Yes, you were right, when I changed the timestamp to 'realtime' in Hesai driver file this error gone. Thank you for your help, appreciate it.

closed 2023-03-20 by sreejithsreeni