hku-mars/r3live #172: rgb点云着色漂移

shanhaoran, 2023-03-24:
您好，感谢您杰出的贡献！
      我测试的设备为ouster-64 和海康工业相机，在使用您提供的外参标定工具标定相机的外参后，发现点云的着色存在着比较明显的漂移。我的config配置为：
```C++
Lidar_front_end:
   lidar_type: 3   # 1 for Livox-avia, 3 for Ouster-OS1-64
   N_SCANS: 64
   using_raw_point: 1
   point_step: 1
   
r3live_common:
   if_dump_log: 0                   # If recording ESIKF update log. [default = 0]
   record_offline_map: 1            # If recording offline map. [default = 1]
   pub_pt_minimum_views: 3          # Publish points which have been render up to "pub_pt_minimum_views" time. [default = 3]
   minimum_pts_size: 0.01           # The minimum distance for every two points in Global map (unit in meter). [default = 0.01] 
   image_downsample_ratio: 1        # The downsample ratio of the input image. [default = 1]
   estimate_i2c_extrinsic: 1        # If enable estimate the extrinsic between camera and IMU. [default = 1] 
   estimate_intrinsic: 1            # If enable estimate the online intrinsic calibration of the camera lens. [default = 1] 
   maximum_vio_tracked_pts: 600     # The maximum points for tracking. [default = 600]
   append_global_map_point_step: 4  # The point step of append point to global map. [default = 4]

r3live_vio:
   image_width: 1440
   image_height: 1080
   camera_intrinsic:
      [1736.574835116879, 0.0, 889.1695567120595,
      0.0, 1720.3327171351075, 561.1026529195532,
      0.0, 0.0, 1.0 ] 
   camera_dist_coeffs: [-0.049495305220577865, -0.1303240762626971, 0.0011282394340686027, 0.020647162890014507, 0]   #k1, k2, p1, p2, k3
   # Fine extrinsic value. form camera-LiDAR calibration.
   camera_ext_R:
      [-0.0323055,-0.0422072,0.998586,
     -0.999462,0.00701493,-0.0320374,
     -0.00565281,-0.999084,-0.0424111]
   camera_ext_t: [-0.0307503,0.149225,-0.189062] 
   
r3live_lio:        
   lio_update_point_step: 4   # Point step used for LIO update.  
   max_iteration: 2           # Maximum times of LIO esikf.
   lidar_time_delay: 0        # The time-offset between LiDAR and IMU, provided by user. 
   filter_size_corner: 0.30   
   filter_size_surf: 0.30
   filter_size_surf_z: 0.30
   filter_size_map: 0.30 
```
     
[image]
在翻阅GitHub上其他人的问题和仔细阅读代码后发现，r3live中使用的是相机到IMU的坐标，因此我也对imu和相机进行了标定，但是使用imu到相机的外参后，r3live运行一段时间就会崩溃。具体报错为：OpenCV Error: Bad argument (The input arrays should be 2D or 3D point sets)。
请问您能给我一些意见吗？

shanhaoran, 2023-03-24:
我还有一个问题，在mesh重建中如果不需要贴图的话，我是否可以不用图像生成稠密点云，直接使用fast-lio2生成的点云进行网格重建？

closed 2023-04-08 by stale[bot]

yahyahajlaoui, 2024-04-02:
hello, did you solve this problem please ?