hku-mars/r3live #179: Need help exporting mesh to a simulated environment

lty2226262, 2023-05-07:
Firstly, I want to express my gratitude for your great work on the project. I was impressed by the video that showcased the ability of several applications. However, while exploring the project's codebase, I could not find the related codes for connecting to UE4 or Airsim.
I would appreciate it if you could kindly share the approaches for connecting to UE4 or Airsim or provide some tutorials on exporting the mesh to a simulated environment. Your guidance would be highly valuable for me to achieve my goals.
Thank you in advance for your assistance, and I look forward to hearing back from you soon.

zhangshuoneu, 2023-05-08:
I have the same problem.

lty2226262, 2023-05-24:
needs help

zhangshuoneu, 2023-05-24:
We successfully derived a 3D model using meshlab, but the model quality is very low.